OpenI6X/opentx #61: Global/special functions long press enter doesn't open context menu

marianomd, 2021-11-18:
I couldn't open this menu for copying or clearing:
[image]

ajjjjjjjj, 2021-12-28:
Not defined for I6X, this is enabled on TARANIS currently.

marianomd, 2021-12-29:
:P But the capture is from QX7. It's too cumbersome to scroll until you find "---". And what about copy & paste? And inserting could be necessary if the order of execution is important. I will check it, give me some time.

psicho100pro, 2022-01-04:
Could you please also look at the function of the TRIM buttons in logic functions? The TRIM buttons can normally be assigned, but the function does not evaluate them at all.
This means that TRIM is assigned as "pressed" or "not pressed", but the value of the function is still not evaluated. Thank you very much.

PR #251 (2022-05-16): Enable basic special function menu
- Enable basic special function menu: Cancel, Insert, but no copy.
- Disable unsupported volume function.
- Cleanup of unsupported functions.

Closes #61