OpenI6X/opentx #82: About external RF module power on pin signal

whqsz, 2021-12-21:
quote:
// External Module
#define EXTMODULE_PWR_GPIO            GPIOC
#define EXTMODULE_PWR_GPIO_PIN        GPIO_Pin_9  // PC.09

PC9 pad in the definition of the target file is the open pin of the external RF module. The RF module is off or on in the model menu, but the pC9 level does not change in the actual measurement. What's the problem?

ajjjjjjjj, 2021-12-21:
What version are you running?
Power toggling radio after change and ensuring that settings were saved enforces correct pin state?

whqsz, 2021-12-22:
> What version are you running? Power toggling radio after change and ensuring that settings were saved enforces correct pin state?
newest v1.1.0 firmware
Because the target has a function pin defined, it is trying, but it is not sure whether it is valid.

whqsz, 2021-12-22:
Whether the external switch option is off or on, the pC9 solder joint is always low.

I'm not sure whether it's a project bug or my own hardware problem

ajjjjjjjj, 2021-12-22:
It looks like regression because it was working after power toggle.

marianomd, 2021-12-22:
When implemented ppm in I worked around code related to this. I tried not to break anything but seems I did.

ajjjjjjjj, 2021-12-28:
This is fixed but to power off you will need to power toggle the radio.

zenghr2188, 2022-01-04:
Has the problem of pC9 pad switch output not been solved?

ajjjjjjjj, 2022-01-04:
As you see in previous comment it is solved with a small issue - power toggle radio after settings save to force module off state. Powering on works perfect.

zenghr2188, 2022-01-04:
Theoretically, it is realized in this way, but the reality is that pC9 pad has no level change and has always been low level.

ajjjjjjjj, 2022-01-04:
No it was not, please do not share false information. Have you tested on latest master?

ajjjjjjjj, 2022-01-15:
I've tested it on release 1.2.0 - works as described.

closed 2022-01-15 by ajjjjjjjj

zenghr2188, 2022-01-17:
When the external module uses ppm protocol, pC9 can work normally. I use crsf protocol. PC9 cannot work normally and has no high-level output.

抄送: "囂賬\~浩 ***@***.******@***.***&gt;;

ajjjjjjjj, 2022-01-17:
@zenghr2188 this is tested by me and it works. Show how do you connect it.