OpenI6X/opentx #85: Sometimes settings are not stored to eeprom no matter the time passed

ajjjjjjjj, 2021-12-22:
as in title

marianomd, 2021-12-30:
I lost some settings after changing them, but was on a sudden reboot (previous usb glitch), maybe we should go back to the standard OTX write delay.

ajjjjjjjj, 2022-01-25:
Lowered to 3s, but probably we can go 2s in next version, i was too worried and datasheet shows about 1 000 000 write cycles endurance.
ft24c128a datasheet shows that it should be fine at 400k and even 1M. Probably there is some issue in i2c configuration or write sequence that is causing issues.

ajjjjjjjj, 2022-01-31:
Some settings are saved only on Soft shutdown. This should be added to readme. Write dealy is already back at 2s so i think that this can be closed.

closed 2022-01-31 by ajjjjjjjj