OpenI6X/opentx #89: Compilation fails: missing string

wimalopaan, 2021-12-24:
I use the following cmake line:

cmake -DUSB_SERIAL=OFF -DGVARS=NO -DLUA_COMPILER=NO -DMULTIMODULE=NO -DDEBUG=NO -DHELI=OFF -DTRANSLATIONS=DE -DPCB=I6X -DOVERRIDE_CHANNEL_FUNCTION=OF
F -DDISABLE_COMPANION=YES ..

but I get the following error:

/home/lmeier/Projekte/openi6x/opentx/radio/src/telemetry/crossfire.cpp:42:21: error: 'ZSTR_BATT_PERCENT' was not declared in this scope; did you mean 'ZSTR_BATT1_CURRENT'?
42 | {BATTERY_ID, 3, ZSTR_BATT_PERCENT, UNIT_PERCENT, 0},
| ^~~~~~~~~~~~~~~~~
| ZSTR_BATT1_CURRENT
make[2]: *** [radio/src/CMakeFiles/firmware.dir/build.make:370: radio/src/CMakeFiles/firmware.dir/telemetry/crossfire.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:493: radio/src/CMakeFiles/firmware.dir/all] Fehler 2

ajjjjjjjj, 2021-12-24:
I was doing changes with only english translation in mind to speed up development. I guess now we have to keep all the languages in sync.

wimalopaan, 2021-12-24:
Fixed this by myself, only locally for the moment ...

wimalopaan, 2021-12-24:
Here is a patch:

[de.patch.gz](https://github.com/OpenI6X/opentx/files/7774535/de.patch.gz)

ajjjjjjjj, 2021-12-29:
Thanks, I have updated even more.
Can you test it and let me know? https://github.com/OpenI6X/opentx/pull/100

wimalopaan, 2021-12-30:
Looks fine now!

PR #100 (2021-12-30): Updated DE translations and EN translations cleanup
- Updated DE translations 
- EN translations cleanup

Closes #89
closed 2021-12-30 by ajjjjjjjj