OpenI6X/opentx #90: Compilation fails: maybe compiler version mismatch

wimalopaan, 2021-12-24:
I'm trying to compile myself on arch-linux.

I get the following error:

/home/lmeier/Projekte/openi6x/opentx/radio/src/gui/128x64/model_setup.cpp: In function 'void menuModelFailsafe(event_t)':
/home/lmeier/Projekte/openi6x/opentx/radio/src/gui/128x64/model_setup.cpp:1494:99: error: cannot bind packed field 'g_model.ModelData::moduleData[((int)g_moduleIdx)].ModuleData::failsafeChannels[((int)menuVerticalPosition)]' to 'int16_t&' {aka 'short int&'}
1494 |         int16_t & failsafe = g_model.moduleData[g_moduleIdx].failsafeChannels[menuVerticalPosition];
|                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
make[2]: *** [radio/src/CMakeFiles/firmware.dir/build.make:1238: radio/src/CMakeFiles/firmware.dir/gui/128x64/model_setup.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:304: radio/src/CMakeFiles/firmware.dir/all] Fehler 2

The compiler I use:

arm-none-eabi-g++ (Arch Repository) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

wimalopaan, 2021-12-24:
Info: using compiler version 4.7.4 works (looks like same restriction as for OpenTx).

ajjjjjjjj, 2021-12-26:
Exactly, also newer compilers create larger bin file which makes them useless at current state (Probably some additional switches could fix that).

closed 2021-12-26 by ajjjjjjjj