OpenI6X/opentx #91: Preflight Checks: S-Warning

megacct, 2021-12-25:
Could the S-Warning checks be extended to include the other switches and pots?

ajjjjjjjj, 2022-01-17:
Is S-Warning option in Model setup working for you? If no please attach sceens or write  detailed description how to reproduce your issue.

megacct, 2022-01-17:
Hi. The S-Warning is working for Switches A and B. I was wondering if the same checks could be done for switches C and D, and maybe the Pots? It's possible it can already do this and I just don't have the config setting correct.

Here is my current setting (default I believe):
[image]
With SW A & B down at startup I get:
[image]
With SW A down I get this:
[image]
With SW B down I get this:
[image]
With both switches UP there is no warning.

jakeeri, 2022-01-17:
I have exact same behavior.
In setup S-Warning row is somehow garbled. On TaranisQ that row show like in attached picture.
[image]

ajjjjjjjj, 2022-01-17:
Thanks for that screen! This will make backporting easier.

jakeeri, 2022-01-17:
Here's the link to Painles360 video on how the switch warnings work: https://www.youtube.com/watch?v=qP1OE5JQpzU

ajjjjjjjj, 2022-02-17:
Are you ok with just switches but without pots?

megacct, 2022-02-18:
Personally, I'd like both but, if I had to choose, I'd rather have switches than pots. Great work by the way. Thanks

tn2009, 2022-02-18:
Just switches. 
Very very nice firmware:)

jakeeri, 2022-02-18:
Switches is enough.