OpenI6X/opentx #94: Use compilation-flag -DCROSSFIRE=OFF to exclude crsf

wimalopaan, 2021-12-26:
... may this will save some flash space

ajjjjjjjj, 2021-12-26:
But we use CRSF for ExpressLRS and on other builds FLASH size is not a problem currently or you mean something else?

wimalopaan, 2021-12-26:
Well, in #90 bin size (flash) is an issue ...

Just make it optional to save space if only AFHDS2A is needed.
Maybe this gives room for other RF protocols on this chipset.

ajjjjjjjj, 2021-12-26:
We don't have a ready feature that is not enabled on standard builds because of flash size. With disabled CRSF on standard build currently you will gain nothing but loose support for TBS/ExpressLRS modules with display that do not require configuration script.
A7105 can only implement Hubsan X4 protocol besides AFHDS2A, but i doubt that someone will implement it.

closed 2021-12-26 by ajjjjjjjj