OpenI6X/opentx #95: PWM does not work with FS-IA6B when in PWM/SBUS mode

tmcadam, 2021-12-26:
**Describe the bug**
When in PWM/SBUS mode, when using at FS-IA6B receiver, PWM is not being output from the receiver.

**To Reproduce**
Install  OpenI6X v1.1.0 on FS-I6X
Bind with FS-IA6B (with FW updated to 1.6....SBUS enabled)
Monitor SBUS output (i.e connect to flight controller or use an oscilloscope)
Monitor PWM output (i.e. connect a servo)

**Expected behavior**
Using an FS-IA6B receiver with upgraded firmware it should be possible to output SBUS instead of IBUS. At the same time it should be possible to output PWM on all 6 channels of the receiver. When in in PWM/SBUS mode, the SBUS serial output works correctly but the PWM iis not working. 

This setup was working previously with the stock FS-I6X firmware. 

In PWM/IBUS mode of  the behavior is correct, with both IBUS and the 6 PWM channels being output from the FS-IA6B at the same time.

[image]

wimalopaan, 2021-12-28:
Please have a look if all 16 Channels for SBUS/IBUS are transferred.
MPM had as issue only tranferring 14 channels for AFHDS2A.
Maybe you can get thta for IBus all 18(!) Channels are transferred.

tmcadam, 2022-01-13:
I tested with the second FS-IA6B last night and seeing the same issue.

PWM/IBUS - OK
PPM/IBUS - OK
PWM/SBUS - Outputs PPM and SBUS signals
PPM/SBUS - OK

This is using the revised receiver firmware with sbus enabled and the rssi output on channel 14. https://github.com/Cleric-K/FlySkyRxFirmwareRssiMod

tmcadam, 2022-01-14:
I did some more debug today and hardcoded the PWM and SBUS settings into `radio/src/targets/flysky/AFHDS2A_a7105.cpp` file (lines 173-177 and 185-189). I took out the conditionals and set packet indexes 13 and 21 to bytes 0x00 and and 0xdd respectively......and it worked. I can get SBUS and PWM on the IA6B. 

So it looks like the issue is not with the protocol or receiver but something with the selection of the subtype in the model setup menu. Possibly something around the enum `AFHDS2A_Subtype` or something in the logic in `checkIncDec`.

Looking at the enum I would expect the following values:
 ```
 AFHDS2A_SUBTYPE_FIRST = 0
  AFHDS2A_SUBTYPE_PWM_IBUS = 0
  AFHDS2A_SUBTYPE_PPM_IBUS =1
  AFHDS2A_SUBTYPE_PWM_SBUS = 2
  AFHDS2A_SUBTYPE_PPM_SBUS = 3
  AFHDS2A_SUBTYPE_LAST = 3
```
So to test, I took out the call to `checkIncDec` in `model_setup.cpp` and hardcoded the subtype value like this `g_model.moduleData[INTERNAL_MODULE].subType = 2;`, but I was getting PPM and SBUS at the receiver. 

I really need to work out some debugging to technique, to look at the enum values and the logic in `checkIncDec`. I saw someone mentioned a debug pad. But no idea how that works. Maybe SWD?

tmcadam, 2022-01-15:
Thanks, got debugging lines setup, even drilled a hole in the back of my radio for them. Guessing I might need them again :-). I've found the problem, and tested, it's in the conditionals in `radio/src/targets/flysky/AFHDS2A_a7105.cpp`. I'll make a PR with the fix tomorrow.

closed 2022-01-16 by ajjjjjjjj