OpenI6X/opentx #119: Internal module is not physically powered off when toggling it

ajjjjjjjj, 2022-01-08:
**Describe the bug**
Internal module draws power no matter the on/off state after initial turn on. Stays off only when power toggled with internal module in off state.

**Expected behavior**
Internal module should stop drawing power in off state.

whqsz, 2022-01-09:
Just upgraded to version 1.2 but still can't change the level output of PC9 through the menu

whqsz, 2022-01-09:
Many players hope to use PC9 pin to control the external module power supply EN signal

ajjjjjjjj, 2022-01-09:
Thanks for the report, will take a look

ajjjjjjjj, 2022-01-15:
I have tested it today and it works as described. Can you share schematics/picture how are you using it?

ajjjjjjjj, 2022-01-15:
I've done some more testing regarding internal module and it stays a little higher after toggling ON and OFF, but nothing to worry about.

closed 2022-01-15 by ajjjjjjjj