OpenI6X/opentx #120: DIY-Multiprotocol-TX-Module

height888, 2022-01-10:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
DIY-Multiprotocol-TX-Module   is not yet possible as an external RF module

**Describe alternatives you've considered**

**Additional context**
Add any other context or screenshots about the feature request here.

bigbfromjtown, 2022-01-12:
So I'm ordering a multi-protocol module it will not work? I've already flashed open TX

ajjjjjjjj, 2022-10-10:
This implementation is possible, serial driver examples are ready in aux serial driver and crsf confiruation. This feature need developer with 4in1 module to sync source with current OpenTX master and test it.

ajjjjjjjj, 2025-01-19:
After pulses logic refactor PR merge, MULTI support can be copied from latest OpenTX and should just work.

Just letting know if someone is still interested.

DaVLeO78, 2025-10-18:
Hello Janek! Thank you a lot for your answer! I see a little bit clearer now  : so it's because Multi-telemetry is disable, that i can't make it work, not only because I'm a fool rookie...! I say that because at first i was wondering where to plug the pin5 of the Irx4 (i not sure to have found but I thing it a serial RX). Latter, after many tries of config of my multi module  using config.h with Arduino, I realized that the pin5 was not connected to the MCU of the Irx4 plus it was lacking a jumping solder...  You give me hope, you know i was near to give up! I'm INTERESTED to make it work : you have a VOLUNTEER to test what needs to, for making workable this option, with of course the supervision  of someone more qualified than me : I promise to do my best for the proposal to all of a multi-telemetry working with Fsi6x. Alone It's impossible for me to do that : I know my limit, but I'm ready to push it to get the result! I'm available if you need more information about my hardware and software configuration. Thank you again Janek! Greetings!