OpenI6X/opentx #121: The last field in special functions has no effect

tmcadam, 2022-01-13:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
I am trying to create an alarm from telemetry. I would like an alarm to sound repeatedly when plane battery is below a set voltage. I have set up a logic condition to for this and it is working. I have then set up a special function to play a sound when this logic condition is met. This plays the sound once. I am setting the last field in the special function to 5. I'm not sure what this does but expect that this would either, play the sound 5 times or repeat the sound every 5 seconds (what I would really like). But changing this value has no effect. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set a special function to activate on a switch i.e SA^ 
2. Select Play sound
3. Set the sound to play
4. Set the final field to number i.e. 5
3. Activate the Switch SA
4. Sound will play once when the switch is moved (no repeating)

**Expected behavior**
The sound to repeat continually at 5 second interval if the switch is activated.

**Screenshots**
[image]
[image]

If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version 1.20.0

ajjjjjjjj, 2022-01-13:
Need to investigate if this is fixed in upstream and could be backported.

ajjjjjjjj, 2022-01-13:
This was enabled only for VOICE & AUDIO capable devices but I have enabled it for our sound solution.
Change will be available in next release.

closed 2022-01-13 by ajjjjjjjj

tmcadam, 2022-01-14:
Built and tested. Works good :-)

ajjjjjjjj, 2022-01-14:
Thanks for detailed report and testing!