OpenI6X/opentx #129: USB Joystick does not work on macOS

ebod, 2022-01-17:
OpenI6X 1.2.0
MacBook Pro with macOS Monterey 12.1

After selecting Joystick mode, transmitter appears in System information as OpenTx joystick, but fails to actually work.

From the dmesg log:
DeviceRequest [81 6 2200 0]:0xe00002d6 means that the host tried to request the ReportDescriptor (aka HID_JOYSTICK_ReportDesc) but the requests times out (timeout value is 5 sec)

```
[3193429.421707]: AppleUserUSBHostHIDDevice:0x10003485e new user client
[3193447.715619]: 579239.144044 HS03@14300000: AppleUSBHostPort::enumerateDeviceComplete_block_invoke: enumerated 0x1209/4f54/0200 (FS i6X Joystick) at 12 Mbps
[3193447.784640]: DK: AppleUserUSBHostHIDDevice-0x100034cee using existing server IOUserServer(com.apple.driverkit.AppleUserHIDDrivers-0x100000716)-0x10000078c
[3193447.784671]: AppleUserUSBHostHIDDevice:0x100034cee start (state:0x0)
[3193447.785204]: [IOUserUSBHostHIDDevice.cpp:504][0x100034cee] Open interface: 0x100034cec
[3193452.786099]: [IOUserUSBHostHIDDevice.cpp:636][0x100034cee] DeviceRequest [81 6 2200 0]:0xe00002d6
[3193457.974595]: [IOUserUSBHostHIDDevice.cpp:636][0x100034cee] DeviceRequest [81 6 2200 0]:0xe00002d6
[3193463.150199]: [IOUserUSBHostHIDDevice.cpp:636][0x100034cee] DeviceRequest [81 6 2200 0]:0xe00002d6
[3193463.251345]: [IOUserUSBHostHIDDevice.cpp:642][0x100034cee] No report descriptor
[3193463.251503]: [IOUserUSBHostHIDDevice.cpp:289][0x100034cee] Start(SUPERDISPATCH):e00002bc
[3193463.251520]: [IOUserUSBHostHIDDevice.cpp:383][0x100034cee] Start failed: 0xe00002bc
[3193463.251563]: [AppleUserUSBHostHIDDevice.cpp:77]Start:0xe00002bc
[3193463.251576]: [AppleUserUSBHostHIDDevice.cpp:119][0x100034cee] Start failed: 0xe00002bc
[3193463.251688]: [IOUserUSBHostHIDDevice.cpp:415][0x100034cee] Close interface: 0x100034cec 0x0
[3193463.251727]: [AppleUserUSBHostHIDDevice.cpp:137][0x100034cee] Stop: 0x0
[3193463.251740]: [AppleUserUSBHostHIDDevice.cpp:126][0x100034cee] Start ret: 0xe00002bc
[3193463.251796]: AppleUserUSBHostHIDDevice:0x100034cee IOHIDDevice::Start:0xe00002bc
[3193463.251818]: DK: AppleUserUSBHostHIDDevice-0x100034cee::start(IOUSBHostInterface-0x100034cec) fail
[3193463.251877]: IOUSBHostHIDDevice:0x100034cef start
[3193480.769155]: IOUSBHostHIDDevice:0x100034cef start failed
[3193480.769265]: IOUSBHostHIDDevice:0x100034cef stop
[3193480.769277]: 579272.197704 IOUSBHostHIDDevice: IOUSBHostHIDDevice::start: unable to start IOHIDDevice
```

ajjjjjjjj, 2022-01-17:
Please add more details - do you have joystick mode forced or you select it on popup, how long do you wait on popup before confirmation etc.

ebod, 2022-01-18:
I set USB Mode to Once/Joystick.

Storage mode works on macOS. I tried joystick on Windows and it works there.
I checked if the it appears on https://gamepad-tester.com/ (mac, win) and Liftoff (mac).

alccrl, 2022-01-20:
Hi,
I use open- ia6x 1.2.0 under linux ubuntu 20.04 3, and about 50 % of the time I get errors and controller is not recognized (velocidrone/fpvfreerider) in joystick mode when I plug in the usb cable.

When it works it is like this 
[ 1738.054081] usb 2-1.2: new full-speed USB device number 4 using ehci-pci
[ 1738.164936] usb 2-1.2: New USB device found, idVendor=1209, idProduct=4f54, bcdDevice= 2.00
[ 1738.164955] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1738.164962] usb 2-1.2: Product: FS i6X Joystick
[ 1738.164967] usb 2-1.2: Manufacturer: OpenTX
[ 1738.164971] usb 2-1.2: SerialNumber: 00000000001B

When it does NOT work it is like this:
[ 1738.054081] usb 2-1.2: new full-speed USB device number 4 using ehci-pci
[ 1738.164936] usb 2-1.2: New USB device found, idVendor=1209, idProduct=4f54, bcdDevice= 2.00
[ 1738.164955] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1738.164962] usb 2-1.2: Product: FS i6X Joystick
[ 1738.164967] usb 2-1.2: Manufacturer: OpenTX
[ 1738.164971] usb 2-1.2: SerialNumber: 00000000001B
[ 1758.757024] usbhid 2-1.2:1.0: can't add hid device: -110
[ 1758.757071] usbhid: probe of 2-1.2:1.0 failed with error -110

This happens just usb connected, or after simulator is started (this latter case shown).
Trying again it may work or not.
may be the exact connect sequence is not clear (switch on, -> Once -> Connect) or (Switch on -> Once -> Connect) or whatelse ?

Alfredo)

alccrl, 2022-01-20:
Addendum: Error code may sometime be other than -110

Alfredo

ajjjjjjjj, 2022-02-01:
Can you try using USB Hub and report back?

alccrl, 2022-02-02:
Do You mean a powered hub , or just any ?
I think I will also try with new 1.3 fw, I use 1.2, just to be aligned to last version

ajjjjjjjj, 2022-02-02:
Any hub will do. Versions 1.2.0 and 1.2.0 should behave the same regarding USB connectivity.

ajjjjjjjj, 2022-02-02:
`[  655.112226] input: OpenTX FS i6X Joystick as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4.4/2-1.4.4:1.0/0003:1209:4F54.0004/input/input22
[  655.168696] hid-generic 0003:1209:4F54.0004: input,hidraw3: USB HID v1.11 Gamepad [OpenTX FS i6X Joystick] on usb-0000:00:1d.0-1.4.4/input0`
This looks like it actually started properly.

alccrl, 2022-02-03:
See lines with error -32 and error -110.

I made a few trials, sometime it does succed, sometime it does not.

Alfredo

alccrl, 2022-02-03:
I noticed that in real opentx (Taranis 9 lite) usb ask mode works and it always connects.
May be some fix is necessary in the code. I Dont know if battery level can affect mechanism, or other causes related to "once" mode.

Alfredo

ajjjjjjjj, 2022-02-15:
Can you check if on this build situation has improved? You have to manually factory reset to use it.
[openi6x-usbtest.zip](https://github.com/OpenI6X/opentx/files/8066778/openi6x-usbtest.zip)

alccrl, 2022-02-15:
Thanks.
I'll let You know (24-48 h).

alccrl, 2022-02-16:
First results (win10/linux):
sometime it disconnects (same errors as before) at the beginning, when game starts.
Afterwards it stays stable.
Controller is always recognized as "unknown type", but sticks are usable.
Fpv sky-dive, Uncrashed, Velocidrone, fpvFreeRider all behave same way.
I noticed that on roll calibration it is necessary to act on trim, in spite of recalibration of sticks on radio controller and reset to factory done twice, but this may be an issue on openiax, not on usb.

I am going to test also usb otg on android with fpv apps fpvfreerider and x-quad sim.
>>I will update this chat afterwards.<<

Summary: I do not see much differences than before.
On win10 I used audio system sounds as "dmesg" to follow up what was going up (usb driver in out sounds).

Alfredo

ajjjjjjjj, 2022-02-16:
ok, thanks. I was hoping for some improvement but i guess it's some different issue.

dim13, 2024-06-28:
Reporting from Sonoma 14.5. Also "Connect USB"→"Joystick", seems to be detected:

```
~ % ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 29>
  +-o AppleT8103USBXHCI@00000000  <class AppleT8103USBXHCI, id 0x10000051c, registered, matched, active, busy 0 (149084 ms), retain 56>
  | +-o FS-i6X Joystick@00100000  <class IOUSBHostDevice, id 0x10031e891, registered, matched, active, busy 0 (35321 ms), retain 27>
  +-o AppleT8103USBXHCI@01000000  <class AppleT8103USBXHCI, id 0x1000003e2, registered, matched, active, busy 0 (3275 ms), retain 42>
```

But that's all. No inputs.

dim13, 2024-07-01:
USB Prober reports joystick connection as:
```
Full Speed device @ 1 (0x01100000): .............................................   Composite device: "FS-i6X Joystick"
    Port Information:   0x001a
           Not Captive
           Attached to Root Hub
           External Device
           Connected
           Enabled
    Number Of Endpoints (includes EP0):   
        Total Endpoints for Configuration 1 (current):   2
    Device Descriptor   
        Descriptor Version Number:   0x0200
        Device Class:   0   (Composite)
        Device Subclass:   0
        Device Protocol:   0
        Device MaxPacketSize:   64
        Device VendorID/ProductID:   0x1209/0x4F54   (unknown vendor)
        Device Version Number:   0x0200
        Number of Configurations:   1
        Manufacturer String:   1 "OpenTX"
        Product String:   2 "FS-i6X Joystick"
        Serial Number String:   3 "00000000001B"
    Current configuration:   failed to get configuration
    Configuration Descriptor   
        Length (and contents):   34
            Raw Descriptor (hex)    0000: 09 02 22 00 01 01 00 E0  32 09 04 00 00 01 03 00  
            Raw Descriptor (hex)    0010: 00 00 09 21 11 01 00 01  22 38 00 07 05 81 03 12  
            Raw Descriptor (hex)    0020: 00 02 
        Number of Interfaces:   1
        Configuration Value:   1
        Attributes:   0xE0 (self-powered, remote wakeup)
        MaxPower:   100 mA
        Interface #0 - HID   
            Alternate Setting   0
            Number of Endpoints   1
            Interface Class:   3   (HID)
            Interface Subclass;   0
            Interface Protocol:   0
            HID Descriptor   
                Descriptor Version Number:   0x0111
                Country Code:   0
                Descriptor Count:   1
                Descriptor 1   
                    Type:   0x22  (Report Descriptor)
                    Length (interface does not currently exist):   56
            Endpoint 0x81 - Interrupt Input   
                Address:   0x81  (IN)
                Attributes:   0x03  (Interrupt)
                Max Packet Size:   18
                Polling Interval:   2 ms
```

dim13, 2024-07-01:
Same with original firmware (working on Mac):
```
Full Speed device @ 1 (0x01100000): .............................................   Composite device: "Flysky FS-i6XCN"
    Port Information:   0x001a
           Not Captive
           Attached to Root Hub
           External Device
           Connected
           Enabled
    Number Of Endpoints (includes EP0):   
        Total Endpoints for Configuration 1 (current):   2
    Device Descriptor   
        Descriptor Version Number:   0x0200
        Device Class:   0   (Composite)
        Device Subclass:   0
        Device Protocol:   0
        Device MaxPacketSize:   64
        Device VendorID/ProductID:   0x284E/0x79A1   (unknown vendor)
        Device Version Number:   0x0100
        Number of Configurations:   1
        Manufacturer String:   1 "Flysky"
        Product String:   2 "Flysky FS-i6XCN"
        Serial Number String:   3 "4F7A4C934130"
    Configuration Descriptor (current config)   
        Length (and contents):   34
            Raw Descriptor (hex)    0000: 09 02 22 00 01 01 00 80  FA 09 04 00 00 01 03 00  
            Raw Descriptor (hex)    0010: 00 00 09 21 11 01 00 01  22 2F 00 07 05 81 03 40  
            Raw Descriptor (hex)    0020: 00 07 
        Number of Interfaces:   1
        Configuration Value:   1
        Attributes:   0x80 (bus-powered)
        MaxPower:   500 mA
        Interface #0 - HID   
            Alternate Setting   0
            Number of Endpoints   1
            Interface Class:   3   (HID)
            Interface Subclass;   0
            Interface Protocol:   0
            HID Descriptor   
                Descriptor Version Number:   0x0111
                Country Code:   0
                Descriptor Count:   1
                Descriptor 1   
                    Type:   0x22  (Report Descriptor)
                    Length (and contents):   47
                        Raw Descriptor (hex)    0000: 05 01 09 04 A1 01 05 01  09 30 09 31 09 32 09 33  
                        Raw Descriptor (hex)    0010: 09 34 09 35 15 81 25 7F  75 08 95 06 81 02 05 09  
                        Raw Descriptor (hex)    0020: 19 01 29 10 15 00 25 01  75 01 95 10 81 02 C0 
                    Parsed Report Descriptor:   
                          Usage Page    (Generic Desktop) 
                          Usage (Joystick)    
                              Collection (Application)    
                                Usage Page    (Generic Desktop) 
                                Usage (X)    
                                Usage (Y)    
                                Usage (Z)    
                                Usage (Rx)    
                                Usage (Ry)    
                                Usage (Rz)    
                                Logical Minimum.........    (-127)  
                                Logical Maximum.........    (127)  
                                Report Size.............    (8)  
                                Report Count............    (6)  
                                Input...................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Bitfield) 
                                Usage Page    (Button) 
                                Usage Minimum...........    (1)  
                                Usage Maximum...........    (16)  
                                Logical Minimum.........    (0)  
                                Logical Maximum.........    (1)  
                                Report Size.............    (1)  
                                Report Count............    (16)  
                                Input...................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Bitfield) 
                              End Collection     
            Endpoint 0x81 - Interrupt Input   
                Address:   0x81  (IN)
                Attributes:   0x03  (Interrupt)
                Max Packet Size:   64
                Polling Interval:   7 ms
```

ajjjjjjjj, 2024-11-19:
Should be working now, if not, leave comment.

closed 2024-11-19 by ajjjjjjjj