OpenI6X/opentx #130: 3 lines are drawn instead of 4

psicho100pro, 2022-01-17:
In the 11th tab of, the custom display menu, 4 row of numbers can be set. But only 3 rows are drawn. The bars work properly.

ajjjjjjjj, 2022-01-17:
4th one is on the bottom inverted bar.

closed 2022-01-17 by ajjjjjjjj

psicho100pro, 2022-01-17:
Thank you for your response.
So in my case, I'm doing something wrong. But I do not know what.
For any combination, I always get only the blinking string 'NO DATA' :
[image]
(

ajjjjjjjj, 2022-01-17:
"NO DATA" means telemetry connection is lost

psicho100pro, 2022-01-24:
I didn't let the problem go to bed. I have a **RX XR602T-A2** through F.PORT.

It never occurred to me to check the RSSI while binning. I couldn't verify that with the original transmitter firmware.
Yes, it's my fault. I never assumed that the RX wouldn't have an RSSI when it sent sensor values ​​to the TX.

The manufacturer provides incomplete or even false information about the product.

Perhaps only one user stated that he managed to get the RSSI into the transmitter. 
https://fpv-racing-forum.de/thread/14466-mamba-f405-mini-xr602t-a-radiomaster-tx16s-kein-rssi-auf-tx/
No one else gets RSSI, I have verified elsewhere.

The manufacturer states this in one of the latest firmware versions in the official documentation.
„Solved the problem of signal interruption of Model A (compatible with FLYSKY
AFHDS-2A) and the excessive jump amplitude of RSSI;
Improve the calculation method of RSSI for OPENTX and part of their radio controller
warning at the lowest required RSSI value; Fixed the checksum and overflow problem of FPORT data. "

The manufacturer also lists some other receiver options, but these are set up in a GUI, which is very difficult to obtain. Again, some options do not work even though this is stated in the documentation.
The manufacturer answers questions, but the manufacturer does not respond at all regarding RSSI.

If this information is useful to you, it is yours. I think that other users will switch like me with this combination of transmitter and receiver.

Thanks again for this port and your excellent work.  Salut