OpenI6X/opentx #131: Add support for i6s radio

ghouet, 2022-01-18:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
The i6s radio has the same stm32 chip. I tried flashing this i6x opentx firmware but it says bad eeprom when starting the radio.
The i6s has an easy way for DFU mode: https://fishpepper.de/2016/09/15/openground-part-2-fs-i6s-dfu-bootloader-access/

**Describe the solution you'd like**
Make this project work with i6s radio.

**Describe alternatives you've considered**
-

**Additional context**
-

ajjjjjjjj, 2022-01-18:
We don't have that device. You have to wait for someone with i6S to contribute to the project or take take action by yourself! ;)

DetourFPV, 2022-01-19:
If someone is working on this in Colorado I can provide a loaner i6S for some testing, but I'd like to have it back...preferably running EdgeTX and ExpressLRS.

DaniilVeselov, 2022-01-21:
https://github.com/fishpepper/OpenGround

Maybe it will be helpfull

ajjjjjjjj, 2024-11-19:
Closing in favor of #386

closed 2024-11-19 by ajjjjjjjj