OpenI6X/opentx #146: BIND button in Flight mode

ajjjjjjjj, 2022-01-30:
**Describe the bug**
After exiting edit mode, moving to next value using BIND button increases field value at the same time.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Flight modes
2. Click on FM0
3. Scroll down to Trims
4. Edit trim value (OK), exit edit mode (CANCEL)
5. Moving to next value using BIND button increases field value at the same time.

**Expected behavior**
After exiting edit mode BIND button should not increase value.

jakeeri, 2022-01-30:
If you select the next/previous value with right pot it doesn't increment the value in the field.
Also if you change the value in the field with left pot, it does change without pressing OK (edit) first.

ajjjjjjjj, 2022-01-30:
Also after moving with BIND it can be corrected using DOWN button, so not critical but it's there.

PR #198 (2022-03-08): Fixed navigation and edit issue (backport)
Closes #146
commit 55864857 (2022-03-08): Fixed navigation and edit issue (backport) (#198)

Closes #146
closed 2022-03-08 by ajjjjjjjj