OpenI6X/opentx #150: AFHDS support is missing in internal RF menu.

rakesh-i, 2022-02-07:
Stock firmware supports older AFHDS receivers but not able to select the AFHDS mode in internal RF menu. Only supports AFHDS2A receivers.(version 1.3.0)

marianomd, 2022-02-21:
I'm working on this. #168

rakesh-i, 2022-02-21:
Thank you! So much. I few older receivers laying around.

Dronard, 2022-05-12:
Hello,
I use openi6x_1.5.0_fr.bin, and I have only FS2A mode or OFF, how to get AFHDS?

Thank for reply

rakesh-i, 2022-05-12:
Implementation is in progress. Check project>TODO>in progress

Dronard, 2022-05-12:
But It's write since 1.4.0 AFHDS is compatible... but not on 1.5.0?

rakesh-i, 2022-05-12:
I have 1.4.0 it doesn't support afhds. What build you are using?

Dronard, 2022-05-12:
[image]

I want to use this

rakesh-i, 2022-05-12:
> [image]
> 
> I want to use this

Yes me too. That's why I opened this thread. marianomd is trying to implement it. I'm waiting for it to complete. Check projects tab for progress

Dronard, 2022-05-12:
Okay!
I don't go often on github, If you can reply here a message like AFHD is ready, I will thank you))

rakesh-i, 2022-05-12:
> Okay! I don't go often on github, If you can reply here a message like AFHD is ready, I will thank you))

Okay

Dronard, 2022-06-14:
Hello ! Does the 1.6.1 version support AFHDS?

rakesh-i, 2022-06-14:
Haven't tried it yet. Most probably not

marianomd, 2022-06-15:
Not yet.

Dronard, 2022-08-08:
Hello!
Any news about it? should we expect to get it on the 1.7.0 ?

Dronard, 2022-09-23:
is it solve?

Dronard, 2022-10-17:
so? have you got an idea of... when?

marianomd, 2022-10-17:
I have no time for this, sorry. If someone is willing to take over it, it's here: #168

Dronard, 2022-11-08:
1.8.0 Millésime is here 👌🏻👍🏻

Dronard, 2022-11-08:
I install 1.8.0 version... I don't have it, could you explain how you do it work?

Dronard, 2022-11-09:
i don't have AFHDS2 mode...

Dronard, 2022-11-09:
[image]
[image]
[image]

Here is what I have in RF binding menu : FS2A or OFF

Dronard, 2022-11-10:
😥 please...

Dronard, 2022-11-12:
oh, i'm wrong i'm wrong, I hope there will be a new firmware which will get afhds compatibility, in addition to afhds2

Dronard, 2023-10-09:
Hello! any news about it?