OpenI6X/opentx #151: sm100 Cable does not work

Ragi42DEV, 2022-02-08:
Hello, I have bought sim cable sm100 and plugged that to my computer and radio and it havent worked. I thaink that the radio does not send any signal to the port so can someone help me with setting that up?

jakeeri, 2022-02-08:
Sim cable doesn't work. You need a cable which came with the transmitter, if it did. Unfortunately that cable is not in every box for some reason.

jakeeri, 2022-02-08:
Data cable for flashing look like this.
[image]

Ragi42DEV, 2022-02-08:
but if i would solder tx2 to s video connector ppm pin and enable external RF to PPM it should work. Shouldn't it?

ajjjjjjjj, 2022-02-08:
What do you want to use that cable? Standard cable (USB) allows for joystick, backup and flashing.

zxh465118755, 2022-02-15:
I seem to have the same problem as you。flysky i6X opentx，PPM Abnormal operation，No export PPM

ajjjjjjjj, 2022-10-07:
So true issue here is PPM output, described currently here: https://github.com/OpenI6X/opentx/issues/283
Therefore I'm closing this one and let's keep PPM discussion there.

closed 2022-10-07 by ajjjjjjjj