OpenI6X/opentx #152: (bug) usb joy stick not working and tx restarts

xpertcrush, 2022-02-14:
**Describe the bug**
When i connect usb as joystick it doesn't work on first go and tx also restarts on connection, on windows after trying 2-3 times it connects but in setup joystick option, none of the axis works, see video for demo.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to setup and change usb to once
2. Connect to pc using data cable
3. It won't connect and you will only see usb input device,
Repeat the steps, your tx may reboot on own and it will connect, but your current model selected will have garbled name and settings inside.
4. See error

**Expected behavior**
It should work on first go and there should be no reboots

**Screenshots**
[image]

[image]

https://youtu.be/cawFnemfd7o

**General information**
 - OpenTX version (e.g. 1.3.0 )
 - Hardware (e.g. Flysky i6x)

ajjjjjjjj, 2022-02-14:
Can you retry on this build?
[openi6x-usbtest.zip](https://github.com/OpenI6X/opentx/files/8062914/openi6x-usbtest.zip)

xpertcrush, 2022-02-15:
Yes i can, wait i will do it now and report back

xpertcrush, 2022-02-15:
> Can you retry on this build? You should manually  format eeprom after flashing it.
> [openi6x-usbtest.zip](https://github.com/OpenI6X/opentx/files/8062914/openi6x-usbtest.zip)
> 
> 

Initially expression:

Looks good so far it didn't rebooted and my sticked detected as well [image]

[image]

Please change batter low range from 3.5

[image]

So i dont hear this continuous beep of low battery on lipo

[image]

xpertcrush, 2022-02-22:
Unfortunately It's back again today

https://youtu.be/elJ7mrpUN3A

marianomd, 2022-02-22:
Should we try disabling the watchdog to see if its a delay or a lockup?

ajjjjjjjj, 2022-03-08:
Can you share your eeprom copy?

ajjjjjjjj, 2022-03-30:
@xpertcrush , maybe i found the fix, try it:
[opentx-i6x-dev-wdg.zip](https://github.com/OpenI6X/opentx/files/8383831/opentx-i6x-dev-wdg.zip)

ajjjjjjjj, 2022-05-05:
Closing because stalled

marianomd, 2022-06-07:
Still happening on 1.6.1. Will try to debug it (through bluetooth serial💪)

marianomd, 2022-06-07:
It seems related to USB cable quality.

marianomd, 2022-06-09:
The throttle warning doesn't appear during the bootloops.

https://user-images.githubusercontent.com/274989/172937311-857c1803-8134-447e-afa5-7d77a5330c04.mp4

.

Pmant, 2022-06-29:
I have a similar issue, never got joystick mode to work. However, flashing via dfu mode and storage mode do work. 
In joystick mode I get a broken device (Code 10) in windows and the i6x randomly beeps a couple times.

througer, 2022-08-24:
Hello, great work so far. The joystick mode does never work for me in version 1.6.0 or 1.6.1. Tried the older versions. Had to flash back to Version 1.1.0. This will do it every 4th or 5th attempt. In very few cases, the device restarts. Most of the time, an error code 10 appears in the device-manager.

DocTrucker, 2022-11-08:
Just updated from the one of the early versions of this that had to be flashed by ST Link Adaptor. Flashed fine by DFU and USB cable on linux. Booted into windows (for simulator) and the Devices and Printers page shows "FS-i6x Joystick" but the hardware tab of the properties shows "Device Status: This device cannot start. (Code 10)"

Version 1.8.0 2022-11-04

woodprop, 2022-12-19:
In version 1.9.0 USB Joystick works fine for me! (in 1.8.0 it was always rebooting). 
only in one case I have my radio rebooted itself - if I set USB mode to joystick, connect the radio to PC, then unplug it and plug in again without powering off the radio. But it is not a big deal...

ajjjjjjjj, 2024-10-07:
Fixed by #424

closed 2024-10-07 by ajjjjjjjj