OpenI6X/opentx #155: openi6X PPM  Not working properly

zxh465118755, 2022-02-15:
**Describe the bug**

I've opened it openi6X External RF  PPM, but openi6X PPM  Not working properly.
PPM-out and TX2 welded on the motherboard have been tried。
There is no signal output。

Moreover, when I have two models, if I switch from mode 1 to mode 2, I will still be in mode 1 after shutdown and restart. If I want to use mode 2, I can only switch from mode 1 to mode 2 and then switch to mode 1 again. Shutdown and restart at this time is mode 2

**Screenshots**
[image]

[image]

[image]
[image]
[image]
[image]
[image]

General information

OpenTX version (openi6x-1.3.0 )
Hardware (Flysky i6x)

tn2009, 2022-02-15:
confirm. changes are not saved immediately.

zxh465118755, 2022-02-15:
The switching model can figure out how to operate, but the PPM output still doesn't work. The PPM output can be used only when the model without the PPM mode is switched back to the PPM model. If the model with the PPM mode is directly switched on, PPM has no response

zxh465118755, 2022-02-15:
I found some rules. If I turn off the PPM, press the return button for a long time to turn it off, and then turn on the PPM mode after restarting, the remote control will have ppm out. However, if I turn off and restart again, even if the ppm is on, there is still no ppm out

xpertcrush, 2022-02-15:
> I apologize for the off-topic question, but 7.9V? what source to power?

2s li ion/lipo

tn2009, 2022-02-15:
> > I apologize for the off-topic question, but 7.9V? what source to power?
> 
> 2s li ion/lipo

a maximum of 6 volts will otherwise burn out.

ajjjjjjjj, 2022-02-17:
@zxh465118755 , do you read readme on main page and/or release notes on how to be sure that settings are stored?

rakesh-i, 2022-02-18:
> > > I apologize for the off-topic question, but 7.9V? what source to power?
> > 
> > 
> > 2s li ion/lipo
> 
> a maximum of 6 volts will otherwise burn out.
[image]
I have been using 2 800mah 14500 li-ion batteries for a month now. The balance connector to charge them.
[image]
This is the voltage range i have set.
[image]
Low battery voltage. I'm getting 4 and a half hours of flying and 6 hours of simulator with internal RF turned off.  
[image]
Best modification i have made so far. There is a voltage regulator and voltage divider on the input side  so, no worry of burning the chip. Don't do this on fs-i6 as it does not have voltage divider which supports higher voltages. (try it at you own risk)

zxh465118755, 2022-02-18:
> @zxh465118755 , do you read readme on main page and/or release notes on how to be sure that settings are stored?

[](https://fanyi.baidu.com/###)[](javascript:void(0);)很抱歉我没有注意到说明
I'm sorry I didn't notice the release notes，I used the correct shutdown method and the problem was solved，However, there are still problems with ppm output. You can only turn on ppm without turning on ppm, and then turn on ppm. Ppmout will not work normally after restart，Sorry again for my blindness。Sorry again for my stupidity

Andikaluku, 2022-02-26:
Bad ppm

PR #311 (2022-11-24): Fixed PPM Out
Closes #155 
Closes #283
closed 2022-11-24 by ajjjjjjjj