OpenI6X/opentx #157: Expresslrs menu stuck, cannot get out

xpertcrush, 2022-02-16:
When i enter the first menu which is for expresslrs, there is no way to get out of it.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'press and hold bind button to goto menu'
2. Click on 'expresslrs'
3. Scroll down to 'to whatever option you want'
4. And you will be stuck there cannot go back to main screen, no matter what button you press.
**Expected behavior**
It should come out if i press cancel button.

**Screenshots**
https://youtu.be/-BEw9rSPoSA

**General information**
 - OpenTX version 1.3.0 elrs
 - Hardware flysky i6x
 Es24tx with latest expresslrs 2.2

ajjjjjjjj, 2022-02-16:
Try long press CANCEL

closed 2022-02-16 by ajjjjjjjj

xpertcrush, 2022-02-16:
Long press works, how to upload Lua which came with firmware I compiled?

xpertcrush, 2022-02-16:
> Lua is not supported. elrsv2.lua is ported to C.

Can i incorporate the kya into code? Which files need edit?

ajjjjjjjj, 2022-02-16:
Not on this device.