OpenI6X/opentx #158: Telemetry menu and display

tn2009, 2022-02-16:
A choice of 8 sensors is available in the telemetry menu. There are only 6 sensors on the screen.

[image]
[image]
 Firmware 1.3.0

megacct, 2022-02-16:
values for 7 and 8 will appear where 'NO DATA' is in your screenshot when data is available

tn2009, 2022-02-16:
> values for 7 and 8 will appear where 'NO DATA' is in your screenshot when data is available
Only 6.
[image]

megacct, 2022-02-16:
What sources have you set for items 7 and 8?

megacct, 2022-02-16:
Certainly works in version 1.1.1
[image]
[image]

tn2009, 2022-02-16:
Strange, restart and it working:)

closed 2022-02-16 by tn2009