OpenI6X/opentx #165: Backport of pre-flight POT checks - low priority

megacct, 2022-02-19:
Thanks for addressing the pre-flight checks for switches. 

If Space and Time permit, please backport the pre-flight checks for POTs. This is low priority and should give way to any other demand; for Space on the device, or your Time.

ajjjjjjjj, 2022-02-19:
No, way for elrsv2 build, it uses too much flash space and i don't want too much fragmentation between builds. So for now this is not possible.
But can wait for better times, when more flash space will be recovered.

megacct, 2022-02-19:
Understood and expected. Can we please keep this open until such time as it's possible?

megacct, 2022-12-20:
_"But can wait for better times, when more flash space will be recovered."_

Hi, could this change squeeze into some of the freespace now? Thanks

ajjjjjjjj, 2023-01-28:
@megacct, unfortunately not with all languages and besides that there is still analogs debug menu that could be also reenabled.

closed 2023-01-28 by megacct