OpenI6X/opentx #170: Issue with the switches

xpertcrush, 2022-02-21:
When you set up your model after a while you lose your switches on the display page as well as they don't react to your input when you try to arm or do other things with switches. few days ago I had this similar issue but I thought maybe I messed up something but today I face the issue again even after resetting the eeprom. It was working fine until yesterday I turned it off but today when I turn it on again and try to arm my quad using the switches it didn't respond so I checked the display and I found there is no switches anymore in input switches are set correctly and even on mixer switches are set correctly using the inputs but if I try to change input values in the mixture to switches I don't see any switches there if I reset the eepROM then I will definitely find the switches again
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

https://youtu.be/_lcqsMPcbrg

**General information**
 - OpenTX version (1.3.0)

ajjjjjjjj, 2022-02-21:
Please describe reproduction steps. What have you changed in your setup then how do you shutdown - was it soft shutdown before powering, etc.

xpertcrush, 2022-02-21:
Yes it was soft shutdown yesterday and i just used it for 20 minutes for flying, no settings were changed. Do not know the step to reproduce, it just happened twice to me. It happened randomly, first time it was in a day, today it happened after working fine for 3 days.

tn2009, 2022-02-21:
I had this after restoring EEPROM.bin

ajjjjjjjj, 2022-03-04:
Please add as much detail as you can, were you using USB / ELRS meanwhile? Which build exactly you are using? etc. Try to create reproduction scenario.

ajjjjjjjj, 2022-03-19:
Please try this build and report back:
[opentx-i6x-dev-eeprom.zip](https://github.com/OpenI6X/opentx/files/8310019/opentx-i6x-dev-eeprom.zip)

ajjjjjjjj, 2022-05-05:
Closing because stalled.

closed 2022-05-05 by ajjjjjjjj