OpenI6X/opentx #171: SBUS-Output on FS-iA10B: missing ch15/16 and weird value on ch14

wimalopaan, 2022-02-25:
Release: 1.3.0

I experience the following problem:

if I use PWM,SBUS and bind to an FS-iA10B rx I get only value 0 (mid) on ch15 and ch16 output and -100% value on ch14. 
I used an RIMLec Sbus/PWM-decoder.
The channel monitor shows the correct values for the channels.

ajjjjjjjj, 2022-02-26:
In SBUS/IBUS there are 14 channels, where the last one contains LQI.
Full 16 channel modes are not implemented (SBUS16, IBUS16), those may be incompatible with some older receivers and will cost some flash space.

wimalopaan, 2022-02-26:
Well, mapping LQI into CH14 for SBus sounds ab bit strange to me.

If I understand you correctly, your intend is to keep the status quo? Or would it be an option to make it a compile-time option?

ajjjjjjjj, 2022-02-26:
I checekd in the origin of the implementation - Multiprotocol repo and on IBUS16/SBUS16 model it is injected to channel 17. On Erfly6  (standard 14ch) it is optional and guarded by define. I think that this should also be changed here to define to free up 14th channel.

wimalopaan, 2022-02-26:
Yes, for MPM LQ is on CH17 for IBUS16, and SBUS16 are plain ch1-16.
Do think this would be an option to implement that for openix6 as well?

ajjjjjjjj, 2022-02-26:
Is making channel 14 a standard channel (without LQI) adding any value to you?

wimalopaan, 2022-02-26:
Well, I made with Pascal the change to Sbus16/IBus16 in the mpm-codebase. And after that I am used to it now ...
So my answer would be: no

But I clearly understand that b) is more effort than a)

wimalopaan, 2022-02-26:
Ok, I'll then do a fork. 
Thank you!

closed 2022-03-08 by ajjjjjjjj

ajjjjjjjj, 2022-03-09:
Now the new 16 channel modes are selectable from subprotocol menu.