OpenI6X/opentx #176: Possibility to complete disable CROSSFIRE / ELRS V2

wimalopaan, 2022-02-27:
To rediuce the flash and RAM size it should be possible to completely disable these two features.

I think this would be a good feature for user that do use the radios in a classical way.

ajjjjjjjj, 2022-02-28:
Elrs is already isolated, don't know about crossfire for sure.
What do you want that space to be used for?

wimalopaan, 2022-02-28:
As I understand, ELRS need CRSF. I just found, that deactivating `PCBI6X_ELRSV2` does not remove the calls to the elrs/crsf stuff. So removing `elrsV2.cpp` and `crossfire.cpp`  leads to undefined references.

The idea behind removing the support for ext. module was just, that there is no ext. module bay. So using an external module need some (minor) HW mod (at least find the pins), and that makes it the same king of mod as soldering an BT module. 
I have some code I want to use for a simple protocol for the HolyBro SikV3 Modem.

ajjjjjjjj, 2022-02-28:
PCBI6X_ELRSV2 removes all elrsv2 hooks for sure. CROSSFIRE define should also work. You may want to disable PCBI6X_USB_MSD which is also very resource consuming + remove declaration of uint8_t MSC_BOT_Data[MSC_MEDIA_PACKET] to save 0,5KB RAM>

wimalopaan, 2022-03-01:
Maybe you might habe a look a #178

closed 2024-10-23 by wimalopaan