OpenI6X/opentx #180: Switching the menu page back while holding the bind button

tn2009, 2022-03-01:
Is it possible to disable scrolling through the menu with the pots. And add the ability to go to the previous menu page while holding the bind button? 
That would be great!

ajjjjjjjj, 2022-03-02:
Sure and AFAIK someone is already working on it.

tn2009, 2022-03-03:
can you do it? :) I think you can do it faster.

ajjjjjjjj, 2022-03-04:
I was curious how to fix this, so i made this change :)

closed 2022-03-04 by ajjjjjjjj

psicho100pro, 2022-03-05:
After this modification in modelsel, the iteration skips the second setup tab directly on the flight modes. 
And if you hold BIND immediately on modelsel, the tables will flash back and forth.
To get to it, I have to hold down the BIND button and jump back from flight modes , otherwise I won't get to the second tab.

ajjjjjjjj, 2022-03-05:
Try now

PR #187 (2022-03-04): Navigate back on long press BIND button.
- Navigate back on long press BIND button.
- Disable POT2 navigation.

I really like this change because as a nice side effect it frees up 130 bytes.
I leave it for discussion/review for a while.

Closes #180
PR #189 (2022-03-05): Fixed an issue where moving right in model menu will skip second tab
Fixed an issue where moving right in model menu will skip second tab.

Closes #180