OpenI6X/opentx #181: Hardware documentation

wimalopaan, 2022-03-02:
The other day I stumbled across kind of HW documentation: it was a table describing the connection of the µC pins.

Unfortunately I can't find that ressource anymore.

Maybe one can link that into the wiki?

ajjjjjjjj, 2022-03-02:
Was it this one? 
[stmicroelectronics-stm32f072c8t7-datasheets-0135.pdf](https://github.com/OpenI6X/opentx/files/8169629/stmicroelectronics-stm32f072c8t7-datasheets-0135.pdf)

wimalopaan, 2022-03-02:
No, I mean a Excel or OO sheet with the connection on the i6x mainboard and some explanation.

wimalopaan, 2022-03-02:
I'm noit familiar with the F0 series. But in `common/arm/stm32/aux_serial_driver.cpp` the DMA is only activated it it is not a `F0` series MCU. The datasheet list a 7-channel dma controller. So: are all channels used? This kind of docu I'm looking for also.

ajjjjjjjj, 2022-03-02:
I have only this (shared by @marianomd )

-------------
DMA
-------------

Channel 2: TX1, TX3
Channel 3: RX1, RX3
Channel 4: TX1, TX2
Channel 5: RX1, RX2
Channel 6: RX2, RX3, RX4
Channel 7: TX2, TX3, TX4

-------------------------------------
Exposed PADS
-------------------------------------

PB8 TIM16_CH1, I2C1_SCL
PB9 TIM17_CH1, I2C1_SDA

USARTS

PA9   USART1_TX
PA10  USART1_RX

PD5  USART2_TX
PA15 USART2_RX

wimalopaan, 2022-03-03:
Ok, that's a beginning!

Is there a survey regarding the timers? (I need a 1ms timer like in EdgeTx)

ajjjjjjjj, 2022-03-03:
Timers are defined in board.cpp: 2MHz and 5ms one. In EdgeTX 1ms is only for hall gimballs pulling, i wonder why is it needed here? Is functionality you're trying to implement already in EdgeTX/OpenTX to take a look?