OpenI6X/opentx #204: Cannot flash to OpenI6X V1.0.0 or above

ardianet, 2022-03-12:
Hi,
I cannot flash my i6X with V1.0.0 release or above.
Last .bin file that worked was marked as : **opentx-i6x-crsfshot-v2** which I got from Telegram (by Mariano on Nov 5th 2021).

My flashing method is using **STLink v2 with STM32 utility (CLK-DAT-GND)**.
Steps that I already done :
1. Erase chip -> flash V1.0.0 (have try all .bin in the Assets) -> stop at "EEPROM Warning" (same goes with other release).
2. Erase chip -> flash with opentx-i6x-crsfshot-v2 -> Factory Reset -> Erase Chip -> Flash v1.0.0 to latest all stop at "EEPROM Warning".
3. Erase chip -> flash with the original firmware -> Factory Reset ->Erase Chip -> Flash v1.0.0 to latest all stop at "EEPROM Warning".

also try with DFU method (bridging R53) with the same result as above.

[image]

My i6x board is with green silkscreen(since I saw the blue one) marked : **FS-i6x-M6-V1.6** and date **2017/12/12**
MCU : **STM32F072**

megacct, 2022-03-12:
have you tried using the STM32CubeProgrammer as per wiki https://github.com/OpenI6X/opentx/wiki

ajjjjjjjj, 2022-03-12:
Can you describe what "(...) all stop at "EEPROM Warning"" means?

ardianet, 2022-03-12:
@megacct I'm using STLINK because I didn't have the flysky stock USB cable. 
Try it with STM32cubeprogrammer (using STlink), once I can get it installled with v1.1.1-RC1. 
Do the factory reset, reflash it to v1.3.0. And the same problem appears. Either it show blank screen or stop at EEPROM Warning screen.

@ajjjjjjjj Right after flashing the firmware, there was the "Bad EEProm data - press any key" screen. And when I pressed it, it will say "formatting the eeprom", and I could say the screen freezed up (no response). Have to cycle the power, but then giving blank screen. 

Will try the .bin.

ardianet, 2022-03-12:
The problem I have is about the same like on this post

ajjjjjjjj, 2022-03-19:
Please try this build and report back:
[opentx-i6x-dev-eeprom.zip](https://github.com/OpenI6X/opentx/files/8310018/opentx-i6x-dev-eeprom.zip)

ardianet, 2022-03-20:
> @ardianet, please try this build and report back: [opentx-i6x-dev-eeprom.zip](https://github.com/OpenI6X/opentx/files/8310018/opentx-i6x-dev-eeprom.zip)

Hi, flashed it with the .bin above still got the same problem.

If I got the "EEPROM WARNING" screen, then most of it will not work :
[image]

But if the home screen displayed after flashing, then it will work :
[image]

*****
So then I tried flashing with random openi6x .bin files (including the old files)
when the flash succeed then I'll do Factory Reset and flashed it with the version above it.

Here's some of the version that works :
[image]

[image]

[image]

[image]

I guess the latest version that works is v1.1.1-RC1

But again.. for the v1.0.0 above it's so random, sometime it flashed successfully.. but most of the time its not.

ajjjjjjjj, 2022-03-20:
> If I got the "EEPROM WARNING" screen, then most of it will not work :
> But if the home screen displayed after flashing, then it will work :

What do you mena by that? What is happening after you see EEprom warning and press a key to format it?

ardianet, 2022-03-20:
> > If I got the "EEPROM WARNING" screen, then most of it will not work :
> > But if the home screen displayed after flashing, then it will work :
> 
> What do you mean by that? What is happening after you see EEprom warning and press a key to format it - is it still freezing?

Yes. Its still freezing.

ajjjjjjjj, 2022-03-20:
I have checked changes between versions and there is no justification in code changes that would break something on later versions, especially when you list 1.1.0 as working.

How long do you wait on "formatting..." screen to know that it is frozen?

Ad.3. This is not true. EEPROM Warning is a standard condition with new eeprom data version and does not mean that something went wrong.

ardianet, 2022-03-20:
> I have checked changes between versions and there is no justification in code changes that would break something on later versions, especially when you list 1.1.0 as working.
> 
> How long do you wait on "formatting..." screen to know that it is frozen?
Lately? about 10 - 20 second.. 
Sometimes more than 30s.. just to test if -maybe- it took that quite time..

> Ad.3. This is not true. EEPROM Warning is a standard condition with new eeprom data version and does not mean that something went wrong.
Well, that's what happen. I'll make a video about it.

ajjjjjjjj, 2022-03-20:
Try this one, it will just skip after 100 failed tries to write block:
[opentx-i6x-dev-eeprom2.zip](https://github.com/OpenI6X/opentx/files/8310981/opentx-i6x-dev-eeprom2.zip)

ardianet, 2022-03-20:
> Try this one, it will just skip after 100 failed tries to write block: [opentx-i6x-dev-eeprom2.zip](https://github.com/OpenI6X/opentx/files/8310981/opentx-i6x-dev-eeprom2.zip)
> 
> I suspect that you eeprom is broken, but hopefuly in area you will never notice.

Already waiting for about 1 minute. Or should more?
https://user-images.githubusercontent.com/39653183/159159386-fc725ac0-4fd1-4d85-930e-ae39b3132e1c.mp4

ardianet, 2022-03-20:
Here's some videos explaining what the problem is

Flashing from Crashed FW to Original FW (Success) :
https://user-images.githubusercontent.com/39653183/159159506-cf2a90d1-54fe-4bdd-a25b-2c5256105772.mp4

Flashing from Full-Chip-Erased state to (old) crsfshot-v2 FW (the Ad.2. Blank screen example)
https://user-images.githubusercontent.com/39653183/159159540-b6d382b0-ed11-4c04-aa6a-4131e44bdc2a.mp4

Flashing from crashed (blank screen) FW to opentx-i6x-dev-eeprom FW (the Ad.1. EEPROM Warning screen example)
https://user-images.githubusercontent.com/39653183/159159583-317fbbed-f9ad-4a24-bee5-c641451d3a25.mp4

_Ad.3. This is not true. EEPROM Warning is a standard condition with new eeprom data version and does not mean that something went wrong._
Successful flashing from crashed FW to crsfshot-v2 FW (the Ad.3. No EEPROM Warning screen, straight to home screen)
https://user-images.githubusercontent.com/39653183/159159679-2d273b79-3dc6-44c6-82d8-1aeecdae785f.mp4

ajjjjjjjj, 2022-03-20:
This one skips also after 100 failed reads, it is not how it should be, but for curiosity, does it work?

[opentx-i6x-dev-eeprom3.zip](https://github.com/OpenI6X/opentx/files/8311454/opentx-i6x-dev-eeprom3.zip)

ardianet, 2022-03-20:
> This one skips also after 100 failed reads, it is not how it should be, but for curiosity, does it work?
> 
> [opentx-i6x-dev-eeprom3.zip](https://github.com/OpenI6X/opentx/files/8311454/opentx-i6x-dev-eeprom3.zip)

No. Still the same.. freezing or blank screen.
Perhaps some setup need to be configured on the STM32cubeprogrammer? (I use STLINK)

ajjjjjjjj, 2022-03-20:
I think that flashing is OK. 
Blank screen means contrast set to 0 - which is consistent with whats going on - invalid value written or read from eeprom.
Same behaviour, no matter i2c driver, this one or previous.

This leads me to conclusion that eeprom, or at least some of the adresses, are broken.
This can be debugged further by trying to write then read every eeprom page (64 bytes) or single byte, but for now i don't want to spend time on issue I cannot reproduce.

Lets keep this issue open until there is confirmation from other users that new driver works or have issues.

ardianet, 2022-03-24:
> Can you try this one? (reverted delays and gpio speed) [opentx-i6x-dev-eeprom4.zip](https://github.com/OpenI6X/opentx/files/8326001/opentx-i6x-dev-eeprom4.zip)

Cannot found the file.

ajjjjjjjj, 2022-03-28:
Weird, it must be a github thing that it deletes files sometimes.I got the new one with absolutely slow timings:
[opentx-i6x-dev-eeprom-slow.zip](https://github.com/OpenI6X/opentx/files/8365404/opentx-i6x-dev-eeprom-slow.zip)

ardianet, 2022-03-29:
No luck. 

I'll try to reach [sossy](https://www.rcgroups.com/forums/member.php?u=900545) to look into this issue, since he got this problem too. Perhaps he could share some of his point of view so developers could narrow the problem.

ajjjjjjjj, 2022-03-29:
I'll prepare at least one more build with values from erfly6.

ajjjjjjjj, 2022-03-29:
I2C timing from erfly6:
[opentx-i6x-dev-eeprom-erfly6.zip](https://github.com/OpenI6X/opentx/files/8370930/opentx-i6x-dev-eeprom-erfly6.zip)

Sossy-Dubois, 2022-03-30:
> No luck.
> 
> I'll try to reach [sossy](https://www.rcgroups.com/forums/member.php?u=900545) to look into this issue, since he got this problem too. Perhaps he could share some of his point of view so developers could narrow the problem.

I gave this another shot today and am yielding the same results as ardianet.

I tried flashing directly to V1.3.0 and the radio booted with the throttle/switch warning but I cannot make it past that screen as it will only recognize the first switch input and then freezes.

I then flashed back to my OEM flysky firmware (which doesn't work, the radio won't even turn on with this) but then when I flash a newer version of OpenI6X ( [opentx-i6x-dev-eeprom-slow.zip] ), I make it to the EEPROM warning screen, press the button to proceed in resetting the EEPROM and then my radio freezes on this screen. I've left it here for over 5 minutes now with no response. After rebooting the radio, I only have a blank screen.

My trim tab buttons are responding with their normal sound and the screen backlight is on so it may be the screen contrast issue that I'm having.

Sossy-Dubois, 2022-03-30:
Update: 

I left my radio turned on with the blank screen and it started chirping at me. I think it was the idle radio alarm.

I'm wondering if I could blindly navigate to the screen contrast settings somehow and try to adjust them..

ajjjjjjjj, 2022-03-30:
@Sossy-Dubois , could you try again with the [opentx-i6x-dev-eeprom-erfly6.zip](https://github.com/OpenI6X/opentx/files/8370930/opentx-i6x-dev-eeprom-erfly6.zip)?

@ardianet , was the fail after eeprom format on [opentx-i6x-dev-eeprom-erfly6.zip](https://github.com/OpenI6X/opentx/files/8370930/opentx-i6x-dev-eeprom-erfly6.zip)?

ardianet, 2022-03-30:
> Update:
> 
> I left my radio turned on with the blank screen and it started chirping at me. I think it was the idle radio alarm.
> 
> I'm wondering if I could blindly navigate to the screen contrast settings somehow and try to adjust them..

I tried this too.. when I got the old version working, I memorize the steps.. but still no luck (but now I remember.. the steps I took is to reach contrast menu.. not the eeprom format.. lol)

ardianet, 2022-03-30:
> @Sossy-Dubois , @ardianet , try this one.
> 
> It has forced lcd contrast to let you navigate thru menu to do factory reset [opentx-i6x-dev-contrast.zip](https://github.com/OpenI6X/opentx/files/8382463/opentx-i6x-dev-contrast.zip) .

I'm still having the same issue..

ajjjjjjjj, 2022-03-30:
Do you have a multimeter and can measure resistors near eeprom ic?

Sossy-Dubois, 2022-03-30:
I just gave this a shot and I'm getting the same results as well. Freezes while resetting EEPROM and then blank screen after reboot.

I have a multimeter and can take the measurements if someone can point out to me where to put my probes.

ajjjjjjjj, 2022-03-30:
red is measurement no 1., green is 2.
[image]

Sossy-Dubois, 2022-03-30:
Okay I'm not sure exactly how to read this but with my multimeter set to "20k" I'm getting 2.66 on the reds and 2.65 on the greens. 

Let me know if I goofed that up..

ajjjjjjjj, 2022-03-30:
Great job! :) That are proper values so i still don't know what's wrong.

Sossy-Dubois, 2022-03-30:
Shoot!

Are you in the US? 

I can just donate the radio to you. That way you don't have to work through a dingbat like me.

ajjjjjjjj, 2022-03-30:
Hahah, Poland here :)

Sossy-Dubois, 2022-03-30:
Hmm I think they'll charge me a bit for that...

I'm willing to do whatever tests you can think of

Sossy-Dubois, 2022-03-30:
Thank you for all of the development

ajjjjjjjj, 2022-03-30:
You're welcome.
There is a way to test the issue if you have a USB to UART converter. It requires soldering to PA9 and GND pins.

Sossy-Dubois, 2022-03-30:
I have one of those ST links

ardianet, 2022-03-31:
> You're welcome. There is a way to test the issue if you have a USB to UART converter. It requires soldering to PA9 and GND pins.

I think I have an FTDI232, is it sufficient?

ajjjjjjjj, 2022-04-02:
Let's sort this out.
This is modified firmware forced to do factory reset on every start, and then freeze after it. It is made only for testing purposes.

1. Connect PA9 to RX of FTDI232 and any GND to GND of it. 
2. Open serial console @ 115200. 
3. Turn the radio on and turn it off after about 5 seconds. (If prompted to format eeprom, do it.)
4. Post the logs here.

Do not repeat the process when not necessary to prevent unneeded eeprom wear.

[opentx-i6x-dev-eeprom-format.zip](https://github.com/OpenI6X/opentx/files/8402372/opentx-i6x-dev-eeprom-format.zip)

ardianet, 2022-04-02:
> Let's sort this out. This is modified firmware forced to do factory reset on every start, and then freeze after it. It is made only for testing purposes.
>
> 1. Connect PA9 to RX of FTDI232 and any GND [...]
Will try it tomorrow.

What's the recommended serial monitor app to be used?

ajjjjjjjj, 2022-04-02:
Any will do, on windows i like TeraTerm.

ardianet, 2022-04-02:
> Let's sort this out. This is modified firmware forced to do factory reset on every start, and then freeze after it. It is made only for testing purposes.
> 
> 1. Connect PA9 to RX of FTDI232 and any GND to GND of it.
> 2. Open serial console @ 115200.
> 3. Turn the radio on and turn it off after about 5 seconds. (If prompted to format eeprom, do it.)
> 4. Post the logs here.
> 
> Do not repeat the process when not necessary to prevent unneeded eeprom wear. [opentx-i6x-dev-eeprom-format2.zip](https://github.com/OpenI6X/opentx/files/8402543/opentx-i6x-dev-eeprom-format2.zip)

using the first file, here's the log :
i6X board started :)
RCC->CSR = 0c800000
i2cI
storageEraseAll
eeWB 256,

ajjjjjjjj, 2022-04-03:
Now this one, i try to narrow down at what it fails:
[opentx-i6x-dev-eeprom-format3.zip](https://github.com/OpenI6X/opentx/files/8403955/opentx-i6x-dev-eeprom-format3.zip)

ardianet, 2022-04-03:
> Now this one, i try to narrow down at what it fails: [opentx-i6x-dev-eeprom-format3.zip](https://github.com/OpenI6X/opentx/files/8403955/opentx-i6x-dev-eeprom-format3.zip)

Quite a bit long log, so I put it on text file :
[LOG_3.txt](https://github.com/OpenI6X/opentx/files/8404038/LOG_3.txt)

ajjjjjjjj, 2022-04-03:
Relaxed timeout a bit and added additional logs:
[opentx-i6x-dev-eeprom-format4.zip](https://github.com/OpenI6X/opentx/files/8404052/opentx-i6x-dev-eeprom-format4.zip)

ardianet, 2022-04-03:
> Relaxed timeout a bit and added additional logs: [opentx-i6x-dev-eeprom-format4.zip](https://github.com/OpenI6X/opentx/files/8404052/opentx-i6x-dev-eeprom-format4.zip)

[LOG_4.txt](https://github.com/OpenI6X/opentx/files/8404088/LOG_4.txt)

ajjjjjjjj, 2022-04-03:
Looks very promising, now write test:
[opentx-i6x-dev_write_test.zip](https://github.com/OpenI6X/opentx/files/8404097/opentx-i6x-dev_write_test.zip)

ardianet, 2022-04-03:
> Looks very promising, now write test: [opentx-i6x-dev_write_test.zip](https://github.com/OpenI6X/opentx/files/8404097/opentx-i6x-dev_write_test.zip)
> 
> Are you on telegram group/ discord? This could go faster there.

Yes, I already asked this on Telegram. But I thought it's just my radio doing the weird thing. So by opening this issue here, I'm hoping others with the same problem could participate. And when I see @Sossy-Dubois had the same problem, I guess some other radio may got this error too. :)

btw, here's the write log :
[LOG_WRITE.txt](https://github.com/OpenI6X/opentx/files/8404301/LOG_WRITE.txt)

ajjjjjjjj, 2022-04-03:
Looks weird, it hanged where it shouldnt:
[opentx-i6x-dev_write_test2.zip](https://github.com/OpenI6X/opentx/files/8404332/opentx-i6x-dev_write_test2.zip)

ardianet, 2022-04-03:
> Looks weird, it hanged where it shouldnt: [opentx-i6x-dev_write_test2.zip](https://github.com/OpenI6X/opentx/files/8404332/opentx-i6x-dev_write_test2.zip) then this one: [opentx-i6x-dev_write_test3.zip](https://github.com/OpenI6X/opentx/files/8404363/opentx-i6x-dev_write_test3.zip)

[LOG_WRITE_2.txt](https://github.com/OpenI6X/opentx/files/8405071/LOG_WRITE_2.txt)
[LOG_WRITE_3.txt](https://github.com/OpenI6X/opentx/files/8405072/LOG_WRITE_3.txt)

ajjjjjjjj, 2022-04-03:
Fixed!
Grab working build here: https://app.circleci.com/pipelines/github/OpenI6X/opentx/376/workflows/90eb320b-c856-4cee-83f7-ab74b643bbd4/jobs/587/artifacts

ajjjjjjjj, 2022-04-10:
Hangs on:
`while (A7105_ReadReg(A7105_10_PLL_II) != 0x9E)`

PR #208 (2022-03-24): eeprom i2c driver rework
Motivation behind this is current solution fails for some users and is unable to run at 400k. I also suspect that i2c hangs at times without auto recovery (reinit).

- port i2c driver from Taranis,
- no paging in read mode - as in eeprom spec,
- ACK polling instead of hardcoded wait,
- auto recovery,
- timeout error handling,
- do not use additional RAM for writes,
- show write indicator on main screen.

This should fix #204 and maybe #170.
closed 2022-04-10 by ajjjjjjjj