OpenI6X/opentx #206: Lua script i6x not work after ELRS 2.3.0 update

stanleykwan76, 2022-03-13:
Factory default BetaFPV tx module (1W) Lua script is work fine.
however, when I update to ExpressLRS firmware 2.3.0, FS-i6x Lua not work anymore.

it only show "AUX" and cannot go any function.

stanleykwan76, 2022-03-13:
[image]
before update ELRS 2.3.0 (Factory shipped BetaFPV 1W JR module)

stanleykwan76, 2022-03-13:
[https://youtu.be/GqKbjlN8QGU](url)

After upgrade tonight.
only the LUA script on i6x not work.

ajjjjjjjj, 2022-03-13:
That's what i expected due to increased max packet size. You have to use 2.2.0 or wait for new version.

stanleykwan76, 2022-03-13:
> 

Got is, maybe I down to 2.2.0 and see how's work.
Thanks for your great work!

stanleykwan76, 2022-03-14:
after clam down and think what I did last night......
I remember I swith the dip swith of JR module to "update firmware" position.

and I forgot to swith it back to "Normal Operation"

Let me try tonight and switch it back and check.

ajjjjjjjj, 2022-03-15:
Because this is already fixed on master branch i'm closing this issue.

closed 2022-03-15 by ajjjjjjjj

stanleykwan76, 2022-04-10:
After upgrade 1.4 version, LUA script work again,
Thanks