OpenI6X/opentx #224: Modes don't work

Dronard, 2022-04-06:
I use Mode 1, it don't work

ajjjjjjjj, 2022-04-06:
Mode 1 is swapped elevator with throttle and i see it changed properly on channels monitor.
What is your issue exactly?

Dronard, 2022-04-06:
https://vimeo.com/696693282

Dronard, 2022-04-06:
[image]

Dronard, 2022-04-06:
[image]

ajjjjjjjj, 2022-04-06:
It's not a mode issue but gimbal wires are swapped. Can you make a photo of your gimbal to pcb wiring?

tn2009, 2022-04-06:
Checked mode 1 is working correctly.

Dronard, 2022-04-06:
> It's not a mode issue but it looks more like your gimbal wires are swapped. Can you make a photo of your gimbal to pcb wiring?

The wires are short, i connect them were the should to be plug, with the flysky firmware, it work fine, I don't know why all seems to be inverted.

Dronard, 2022-04-06:
[image]

Dronard, 2022-04-06:
[image]

Dronard, 2022-04-06:
I simply done a strap on R53 only for the first time I upgrade to open TX firmware, then I unplug this one, because I can enter DFU mode by click on the two trim button and power on

ajjjjjjjj, 2022-04-06:
Thanks for the pictures,  can you try to swap left and right gimbal connectors to see if that helps?

Dronard, 2022-04-06:
[image]

Dronard, 2022-04-06:
Here is the tutorial of Rakish RC, were can I find this version ? it may be the solution

Dronard, 2022-04-06:
> Thanks for the pictures. Can you try to swap left and right gimbal connectors(black-white with red-white one) to see if that helps?

Ok

Dronard, 2022-04-06:
Thank you so much, i swap these wires and now everything work fine, I don't know why ancient FS firmware took this configuration right, but now it's good for me. Thank you for your replies and thank you Rakish for your precious tutorials, thanks all for this openTX firmware! I was so much upset by the poor 3 mix on FS firmware...

ajjjjjjjj, 2022-04-06:
Thanks! will pass that to Rakesh, happy flying!

closed 2022-04-06 by ajjjjjjjj

Dronard, 2022-10-11:
Did you receive something ? I send you a video
-------- Message d'origine --------De : Janek ***@***.***> Date : 06/04/2022  18:54  (GMT+01:00) À : OpenI6X/opentx ***@***.***> Cc : Dronard ***@***.***>, Author ***@***.***> Objet : Re: [OpenI6X/opentx] Modes don't work (Issue #224) 
Mode 1 is swapped elevator with throttle and i see it changed properly on channels monitor.
What is your issue exactly?

Dronard, 2022-10-11:
https://vimeo.com/696693282
-------- Message d'origine --------De : Janek ***@***.***> Date : 06/04/2022  18:54  (GMT+01:00) À : OpenI6X/opentx ***@***.***> Cc : Dronard ***@***.***>, Author ***@***.***> Objet : Re: [OpenI6X/opentx] Modes don't work (Issue #224) 
Mode 1 is swapped elevator with throttle and i see it changed properly on channels monitor.
What is your issue exactly?