OpenI6X/opentx #226: Channel monitors display option

megacct, 2022-04-08:
Hi,

Could there be a menu option to select either percent values or PPM on the channel monitors rather than having it as a compile time option?

Thanks
[ great work as always ]

ajjjjjjjj, 2022-04-08:
Unfortunately to keep things simple it's better to not do that, but using PPM_UNIT=PERCENT_PREC1 you will get you one digit of fractional part i.e.: "100.0", is that good enough for you? 
I will consider that change for future version.

marianomd, 2022-04-13:
I also miss the PPM values display. Maybe we could add that variant? This would be handy also when setting up Betaflight modes as the channels are shown in that way:
[image]

ajjjjjjjj, 2022-04-16:
Hmm... :)

When it comes to mixing and overall model setup i think that operation on percent values is easier. Let's say i need to limit elevator to 70% and shift it by 10%. Ii can easily see on the mixers monitor that the value is correct between -60% to 80%. When it comes to switches, for example i want to emulate 6 pos switch by using SWC and SWD: limit SWC to jump over 15% to 30% and 45% and shift it by +50% using SWD to achieve 65%, 80% and 95% positions - no need to additionally map to 1000 - 2000 range.
It is of course only my point of view, but all of the existing radios also use this setup: 100 or 100.0. So it's nice to see the same values on every radio.

Let's keep this discussion open, i don't want to force anyone. Just my point of view.

megacct, 2022-04-16:
I was using the TX to constrain the PWM output to the min/max of 1000/2000, which was why I was disappointed that the new version uses %age. I've now done the constraining at the RX end (arduino) so using percent vals on the TX is nolonger an inconvenience. I'm happy to close this is issue.

ajjjjjjjj, 2022-05-08:
@megacct , @marianomd Please vote: https://github.com/OpenI6X/opentx/issues/226

megacct, 2022-05-08:
Hi Janek, not said often enough, great work. On this one I abstain for the reason mentioned previously. Happy for [@marianomd](https://github.com/marianomd) and yourself to make the call now.

closed 2022-05-09 by ajjjjjjjj