OpenI6X/opentx #246: Blank screen after updating FS-i6X

Mutomatic, 2022-04-30:
Hey,
So i have flashed my FS-i6X, with Openi6x 1.4.0.
And when i power cycle the radio, i just have a blank screen, nothing shows up. Only sign of it being updated, is that in order to into DFU mode to try and flash it again, can be done with the trim buttoms.

I used, STM32CubeProgrammer with the original usb cable through the trainer port.
Everything goes as planned when i flash, no errors (i think).

Any help will be gladly appreciated

Mutomatic, 2022-04-30:
i have now tried a couple of things. tried to reflash back to the original firmware, in hopes of seeing if the LCD is working.
after alot of trial and error, i found out i cant just flash my old Flysky firmware backup. I found some firmware on the internet, with the original bootloader. got it flashed. i can now hear the original sounds of the flysky, but the LCD is still NOT working...

how can i have broken my LCD, i did not even touch it?

if anyone can get it to work with OpenTX, or if anyone can help me just get back to the original firmware of the Flysky i6x. then i would be grateful.

psicho100pro, 2022-04-30:
did you read this? https://www.rcgroups.com/forums/showpost.php?p=48404255&postcount=178

Mutomatic, 2022-04-30:
Thanks for a reply. i have look at his solution, and tried flashing 1.1.1 (which went succesful)
but it did not change anything.

Things to note which is different from me to the video the guy sent.
When he gets his beep, his LCD lights up with the eeprom warning. I can consistently get my beep, but nothing on the LCD, besides the backlighting.

ajjjjjjjj, 2022-05-01:
Hi, how do you power your device?

Mutomatic, 2022-05-01:
Im powering it with a 2s lipo, with a 6 volt voltage regulator.

ajjjjjjjj, 2022-05-01:
Because it looks like damage from too high voltage. Especially when going back to original does not solve issue.

Mutomatic, 2022-05-01:
Would you say 6 volts is to high? just asking. i can also provide it with 5 volts from the voltage regulator.

So is the whole radio just dead?

Mutomatic, 2022-05-01:
i see your comment on the other thread. but he mentions it runs straight off 2s. i bought a 5-6 volt regulator just because i new it could take straight power from the 2s

ajjjjjjjj, 2022-05-01:
6V is OK afaik. That's the max. allowed by internal regulator. But it looks like internal regulator failed for some reason and damaged MCU. You can check regulators with multimeter to see the output voltage to confirm.
Of course i may be wrong.

Mutomatic, 2022-05-01:
Do you know which are the regulators. for example R53 is the pad for DFU mode. where is the regulators if i wanted to check

ajjjjjjjj, 2022-05-01:
[image]

Mutomatic, 2022-05-01:
So you just want me to check the output voltage, from gnd to vout? just making sure.

ajjjjjjjj, 2022-05-01:
It's up to you. I'm pointing out how to check what has failed.

Mutomatic, 2022-05-01:
Yes i know, im happy for the help. so there is 3.3 on both of these from gnd to vout

Mutomatic, 2022-05-01:
and 6.08 volt in

ajjjjjjjj, 2022-05-01:
So looks like everything is fine. Can you try back on original FW?

Mutomatic, 2022-05-01:
Sure, not to be annoying but can you somehow provide me with the firmware .bin
because i feel like im not finding the correct one.
it needs to be able to be flashed with stm32, everything i find, uses the flysky original firmware flasher, which doesnt work.

ajjjjjjjj, 2022-05-01:
Haven't tested it but here it is:
[flysky_i6x_stock.zip](https://github.com/OpenI6X/opentx/files/8599674/flysky_i6x_stock.zip)

Mutomatic, 2022-05-01:
Okay now i flashed it.
So now i have the original sound beeps when the throttle is not all the way down. interesting it wants me to put the stick all the way up (doesnt matter)

 again the trims buttons make it beep. But the LCD still doesnt show anything. only the backlight is on

Mutomatic, 2022-05-01:
So i think it flash succesfully, but the LCD just doesnt show anything.
Can you take a picture of your LCD connector, mines feels like its lose, and it doesnt click in nicely

ajjjjjjjj, 2022-05-01:
I can't be 100% sure, but for me it confirms damaged MCU.

Mutomatic, 2022-05-01:
mmh. okay, that sucks i guess. So it is just dead. and it cant be the LCD that is somehow broken?

ajjjjjjjj, 2022-05-01:
i doubt it. That: 
> (...) interesting it wants me to put the stick all the way up (doesnt matter)
suggest something wrong with MCU ADC.

Mutomatic, 2022-05-01:
Okay, which means what? at this point i feel like it cant be solve, if i understand you correctly.

ajjjjjjjj, 2022-05-01:
Sorry, i meant that it's not LCD, it's MCU that is broken.

Mutomatic, 2022-05-01:
Yes, which is worse, since i cant fix the MCU, i guess.

ajjjjjjjj, 2022-05-04:
Yes, only replacement would help, but with current STM32 prices it is questionable.

closed 2022-05-04 by ajjjjjjjj