OpenI6X/opentx #252: Setting USB-Detect to auto does not work

wimalopaan, 2022-05-16:
Setting USB-Detect to auto, then connecting to PC does not work.

One has to select USB-Mode: auto manually.

ajjjjjjjj, 2022-05-16:
To make "Auto" mode work you need to use `PA15` pad as USB VBUS, preferably thru resistor. FlySky i6X don't have that feature - standard cable have only GND, D+, D- lines.

wimalopaan, 2022-05-17:
Is there a description how to modify the radio? 
So, the standard cable to the PC need to be replaced as well?

ajjjjjjjj, 2022-05-17:
Yes and yes, see main readme.md for all possible connections and approximate location of the pads.

wimalopaan, 2022-05-17:
Where can I get a replacement cable with this strange connector?

So. I have to use a resistor from PA15 to VBUS? Which value? 10k?

ajjjjjjjj, 2022-05-17:
This connector does not provide USB VBUS line, it must be traced from PCB. To make it work you need new usb socket.
10k should be ok, i'm using 1k if i remember correctly.

ajjjjjjjj, 2022-10-12:
Updated readme to make it more clear.

closed 2022-10-12 by ajjjjjjjj