OpenI6X/opentx #253: Save/Restore EEPROM.BIN to PC does not work

wimalopaan, 2022-05-16:
Connecting radio to PC and manually selecting USB-detect auto gives a connection to PC (stroage mode).

One can then copy EEPROM.BIN to PC.

If one tries to restore the EEPROM.BIN (linux: shell command), the file (seems) copied, the radio disconnects immediate, but the it reports an EEProm storage error (or sometimes EEprom full).

ajjjjjjjj, 2022-05-16:
Thanks, i will take a look into that.

wimalopaan, 2022-05-17:
> Thanks, i will take a look into that.

Thanks! (My tests were with 1.5.0)

ajjjjjjjj, 2022-05-17:
I have reproduced it and found horrible bug i have made. It occurs only when copying eeprom using USB. I'm testing fix atm.

ajjjjjjjj, 2022-05-17:
@wimalopaan , can you test it?

wimalopaan, 2022-05-17:
> @wimalopaan , can you test it?

Of course. Did you make a PR?

PR #254 (2022-05-17): Fixed bulk eeprom read.
- Fixed bulk eeprom read.
- Added proper eeprom write wait.
Closes #253
closed 2022-05-17 by ajjjjjjjj