OpenI6X/opentx #256: State of Companion Software

cmorty, 2022-05-22:
I tried to get the companion software to run (I'm new to opentx) and read out the config from my remote. Besides some bugs I had to fix in the build system and the code, it seems to be in an in-between-state at quite a few places. Is there any effort going to support the i6x in the companion software? Is it worth finding a different version of the companion software and use that to add support?

ajjjjjjjj, 2022-05-22:
Hi, there is no ongoing work on companion. I guess the best way would be to work on latest OpenTX branch because current OpenI6X codebase related to it is old and most probably broken.

linuxras, 2023-07-18:
Work has started on a Companion port, stay tuned :)

Growflavor, 2024-07-08:
> Work has started on a Companion port, stay tuned :)

Is there a fork or link we should watch regarding the Companion Port?  Thanks!

Rixxxi, 2025-02-01:
Is there currently a way to configure Openi6X with Companion?
I'm new to this.
I downloaded and installed the newest Companion (v.2.10.5) for Windows x64 from the EdgeTX website. I then connected my i6X with newest Openi6X v2.0.0 to my PC and selected "USB Storage" in the radio. My PC then shows a new drive with the files _EEPROM.BIN, FIRMWARE.BIN_ and _FIRMWARE.TXT_ (empty). So far, so good.
When I open Companion and click "Read Models and Settings from Radio" I get the error message "Unable to find radio SD card!". Am I missing something or is Openi6X just not compatible with Companion?

ajjjjjjjj, 2025-02-01:
Companion support needs to be implemented (adjust data model), no one contributed to it yet.

machcnz, 2026-02-23:
Windows and Linux Releases are up.