OpenI6X/opentx #262: Removal of soft power-off feature disables timers persistence.

marianomd, 2022-05-28:
I'm not finding a way to persist timers after power off, even with the "Save all" option, without the soft poer off function (keeping cancel button pressed)

marianomd, 2022-05-28:
Even if the "save all" function works, it is tedious to have to select that option each time you want to turn off the radio.

ajjjjjjjj, 2022-05-29:
- It was not a true shutdown,
- was wasting flash space,
- animation could be triggered accidentally,
- most of the users don't need it anyway, 
- write indicator solves write delay issue for most cases.

3 button presses vs unusually long delay is not worse from my perspective and from UX standpoint current option clearly describes what it is doing.

marianomd, 2022-05-29:
For me it's a show stopper. 3 button presses every time I turn off the radio is a no go. I suppose that flash space can be reduced without removing the feature. The "unusually long" delay was on purpose, and a standard feature of opentx, and if too much, it could be lowered. Did you make a poll to state that most of the users don't need it anyway?

marianomd, 2022-06-11:
Seems it's just me, will fix the guards and eventually create a new bin in the CI/CD.

jakeeri, 2022-06-12:
No it's not just you. I Just forgot to comment.
I think the same, I'd like to have it as it were or better yet automatic timer saving.

Maybe drone pilots doesn't need or care about the feature but as a airplane pilot I sure like to know how many minutes/hours I have flown with the airframe.
Now it would be easiest to change the menu order so that the save is first then the amount of pressing buttons were reduced to two instead of three. That's not the optimal solution though.

marianomd, 2022-06-14:
Let's reopen then.

jakeeri, 2022-08-12:
@marianomd have you had time to look at this?