OpenI6X/opentx #265: flash to original firmware

jiri-jirus, 2022-06-13:
it is unclear in Flashing Guide whether it is possible to flash back orignal FlySky firmware, and how to do that.

thanks.

marianomd, 2022-06-14:
It's explained under **Alternatively flash using STM32 CubeProgrammer:** 

"First, make a backup of your firmware. Fill in: Address: 0x08000000 Size: 0x20000, then press Read, then Save As..."

marianomd, 2022-06-14:
The "EdgeTX buddy webpage" didn't work for me. It said something like "Invalid device".

jiri-jirus, 2022-07-04:
so this will restore even the original bootloader?

psicho100pro, 2022-07-11:
If you have a backup created before flashing then everything will return to the original.

yuryuoi, 2023-01-10:
If anyone needs the stock firmware, here you go: https://mega.nz/file/6nwUgaKD#VjrGvq5Uw_kmVUQHWH4AbpVrYWwM2e2SmHvVOzWwAqM
Hardware version is V1.0. Of course, it's better if you make your own backup, as I can't guarantee this will work.

AzwadAbid, 2024-01-28:
> stock firmware

@yuryuoi, could you please share your backup file again? I could really use it. Thanks a lot.

Ashutosh07, 2024-05-08:
Hello ,
Is anyone able to share a copy of stock firmware again please? 
thanks

skycraftmike, 2024-05-09:
Hi all 
Just got a FS-i6X and cant wait to start will make backup of stock firmware have done a couple of 9X radios 
mine has the APM32 chip but it looks like it will work with the ST32 Cube going to bridge the R53 and see if I can download the stock firmware will share if i get it 
this is like a new toy you cant put it down and so many options

skycraftmike, 2024-05-09:
hi here is my original firmware file how do i post it to this git cant upload it

Ashutosh07, 2024-05-09:
Hey @skycraftmike ,
Could you share maybe a google drive link for the firmware file? Also just curious if the STM32cube worked with the APM32 chip. As i had the same and I had to use Qmktoolkit to flash.

skycraftmike, 2024-05-09:
did send you email but it says file to big then i just send you the firmware.bin file ST32 does not see my radio i used the dfu-util to flash opentx on it much better than standard firmware more stuff to set and hack was or are looking for a way to pus SD card reader on the radio like i ddi with my 9X radio will search and let know If i get something

skycraftmike, 2024-05-09:
oh forgot to tell you my operating system is linux ubuntu 22 and the STM32cube runs on it but does not pick up my radio

ajjjjjjjj, 2024-11-19:
Added notes to wiki and main readme.md.

closed 2024-11-19 by ajjjjjjjj