OpenI6X/opentx #268: Telemetry Mirror & Baud-rate Change - TX Freeze-Up

FRPProject, 2022-07-03:
**Describe the bug**
When the Telemetry Mirror is ON, Baud rate change from lower values to higher ones causes the TX to freeze. I am using ELRS 2.5.0. 

**General information**
 - OpenTX version (1.6.1 Backlight Mod -en)

ajjjjjjjj, 2022-08-11:
Hi, going from higher one to lower one does not trigger that issue?

FRPProject, 2022-08-11:
yeah, from high to low seems to be okay but the issue happens when baud rate is changed from lower values to higher one.

ajjjjjjjj, 2022-08-29:
@FRPProject , is this happening with BT module or any other UART device connected or it doesnt matter?

FRPProject, 2022-08-29:
when I was testing, my ELRS module was connected to the i6x board, without BT module

ajjjjjjjj, 2022-10-07:
I cannot reproduce this issue, works fine on mine. It sounds weird because even if something bad happen, watchdog should reset radio like it is happening with USB issues.
Maybe there are other circumstances?

FRPProject, 2022-10-07:
Just checked again and it seems the problem is now resolved (I have updated the Betaflight, ELRS, and OpenTX firmware since then though.) Anyways, we can call this issue resolved. Thanks.

closed 2022-10-07 by FRPProject