OpenI6X/opentx #270: Formatting on power on

Bash365, 2022-07-09:
While using my TX on the latest version I put it off for maybe 5 minutes and on powering on was prompted with, "BAD EEPROM DATA" press any key to format. very random.. lost all models and settings.

psicho100pro, 2022-07-11:
If you have an external power supply from 2S, then there is a high probability that your MCU is irreversibly damaged.
Try to return the native firmware and try all the features, then you will see.

Bash365, 2022-07-11:
Im using power from a 2s lithium ion batt which i believe is lower that 2s lipo but im yet to see any thread on such batteries damaging anything. Im still using the transmitter this error only happened once. I highly doubt its because of the voltage supplied. Its a firmware issue.

psicho100pro, 2022-07-11:
MCU damage due to 2S power supply has been confirmed by several people at various functional levels.

Bash365, 2022-07-11:
Your answer is appreciated although it doesnt help. What are other symptoms of a damaged MCU? if my TX is otherwise fully functional other than the one time it prompted a reboot..
I done some searching via google and find nothing on lipos damaging this transmitter, in fact theres people who have been using it this way for years and have yet to report any faults. 
Your suggestion to reinstall the original firmware does not clear anything up either. What is it, "I will see", as you say in your original comment? will I have a non functional radio? the radio is working fine at the moment I still use it almost daily and the error hasnt happend again.

psicho100pro, 2022-07-15:
If we can't reproduce the conditions under which the error occurs, we can't fix it in the code. This is from my point of view, I can only read C, this is how I would solve it in JS.
Just read the errors in the repository regarding this point. As an example I will give https://github.com/OpenI6X/opentx/issues/204 .
One more thing, if there was no shutdown while the settings were being saved, the small square in the upper right.

ajjjjjjjj, 2022-08-11:
As @psicho100pro stated, most probably you have turned off your radio while there was ongoing write to eeprom indicated by square icon in the top right corner of main screen.

closed 2022-08-17 by Bash365