OpenI6X/opentx #275: channels 5 and 6 dont work over the usb trainer cable

fuzzbuzzbay999, 2022-08-28:
I know this is probably a well known issue, but couldn't find any info on why it is a thing, as the original firmware didn't have this issue

--some background--
When the transmitter is plugged into a windows pc, using the trainer, any input mapped to channels 5 and 6 will not affect the Z rotation or sliders, however channels 7 and 8 work

**To Reproduce**
Steps to reproduce the behavior:
1. plug the transmitter in and connect
2. open "setup usb game controller"

whatever is bound to 5 and 6 will not do anything
[image]
[image]
[image]

**General information**
 - OpenTX version 1.6.0
 - FS I6X
- windows 10

fuzzbuzzbay999, 2022-08-30:
What lib is currently being used to create and send the gamepad signals?

ajjjjjjjj, 2022-08-30:
Currently it is using STM32F0x2_USB-FS-Device_Lib_V1.0.0

fuzzbuzzbay999, 2022-08-30:
"radio" is the files used to translate openTX to i6x hardware, and, "companion" is raw openTX. 
Is that correct?

ajjjjjjjj, 2022-08-30:
radio/src contains "everything":
- targets/ - means various boards, ie.: taranis and its variations, 
- targets/common - files common to platform, like stm32 variations: F0, F2, F4,
- thirdparty/ - libraries

USB specific code is in targets/common/stm32.

fuzzbuzzbay999, 2022-08-30:
okie, so i think i have a solution, what if the button count was to be shrunk from 16 to 12. to maybe "open up" the data line, to maybe avoid what could possibly be a buffer overflow? 
It is certainly weird though

fuzzbuzzbay999, 2022-08-30:
Dont know how to word it, but i did knotice only the first 16 indexes of the hid_Buffer were being sent, so maybe changing the button count wouldn't help

ajjjjjjjj, 2022-08-30:
Have fun experimenting on that :) I have remapped one of the channels because sending anything other than 0 at that byte causes complete mess on output data.

Revenger, 2023-07-18:
Any updates on this as I am setting it up to use in sims and not having channels 5-6 is slightly annoying.

EnTune926, 2023-08-29:
> Any updates on this as I am setting it up to use in sims and not having channels 5-6 is slightly annoying.

I would also like to know if any progress has been made here.

ajjjjjjjj, 2024-10-07:
Fixed by #424

closed 2024-10-07 by ajjjjjjjj

fuzzbuzzbay999, 2024-10-08:
yay!!!