OpenI6X/opentx #276: ELRS V3.0 - Compatibility Issue

FRPProject, 2022-09-10:
**Describe the bug**
I have upgraded my ELRS modules from 2.5 to 3.0 and have encounter a few issues with the i6x openTX 1.6.1 interface. 

*To Reproduce**
Steps to reproduce the behavior:
1. Go to ExpressELRS
2. Wait to fully load the menu
3. Connect to your ELRS receiver 
4. See the issue in front of the "Switch Mode" and "Model Match"
5. Go back to the main menu of OpenTx 
6. Your model name is wiped off to "    ,   " and FM clock is activated on its own. 

**Expected behavior**
In the ExpressELRS menu, "Switch Mode" and "Model Match" should represent "Hybrid/Wide" and "ON/OFF". The i6x openTX 1.6.1 should preserve the model name and main menu.

**General information**
 - OpenTX version 1.6.1
 - HappyModel ELRS ES24TX Tx Module - Flashed with ELRS V3.0

ajjjjjjjj, 2022-09-11:
Hi, it;s a known issue, official release needs some tuning, I'm working on it.

FRPProject, 2022-09-11:
Thanks for looking into this issue. I appreciate your work.

closed 2022-09-13 by ajjjjjjjj