OpenI6X/opentx #283: PPM output issues with new version 1.7

RobotSub, 2022-10-04:
I don't think the PPM output is working correctly even though the notes indicated that it was fixed.

Test sequence (repeatable) for FS-I6X
⦁	Load latest factory I6X firmware via DFU / STM Cube programmer (x08000000/x20000)
⦁	Power on
⦁	Confirm correct 8 channel PPM signal on trainer port connector - Yes
⦁	Load OpenTX I6X 1.7 English firmware (openi6x_1.7.0_en_220912.bin) via DFU / STM Cube programmer (x08000000/x20000)
⦁	Power on. Bad EEprom data warning, EEprom formated.
⦁	long press OK, model 01 highlighted
⦁	short press Bind, setup 2/11 displayed
⦁	press down repeatedly to Internal RF Mode
⦁	Ok, down, Internal RF Mode now OFF, click Ok
⦁	press down to External RF Mode
⦁	Ok, Up, now PPM displayed, Ok
⦁	Confirm correct 8 channel PPM signal on trainer port connector - Yes
⦁	cancel, cancel, now at top menu
⦁	confirm update box is not shown (I have also tried this sequence doing a save all)
⦁	At this point I still have a good PPM signal but if I power off and power on there is no PPM output signal
⦁	I now confirm that all of the settings were saved, all ok, also confirmed that trainer mode is Master Jack, still no PPM output
⦁	Go back to External RF and select CRSF, the CRSF serial output (from internal pad) is correct - Yes
⦁	Power off and power on and the CRSF output is still ok - Yes
⦁	Go back and try to adjust settings for PPM, no PPM output.

Notes:  I have also tried with openi6x_1.7.0_elrsv3_en_220912.bin.  To get the PPM output as above it is required that I first load the factory software then load the new software.

ajjjjjjjj, 2022-10-04:
So you have to toggle PPM out off and on to make it working if i understand the issue correctly.

What was the last version it was working correctly for you?

RobotSub, 2022-10-04:
Thank you for the response, I will do additional version testing and report back. Is this the best spot or discord or ?

ajjjjjjjj, 2022-10-04:
It's a bug so it's a perfect place to report it as an issue.

RobotSub, 2022-10-05:
Thanks again for your response. I have checked multiple versions of the code and they act the same so perhaps it's not a bug (?) but I have spent several days playing with this and read everything online I could find so I seem to be at a stopping point , getting to my problem is simple compared to above list, the list below is fully repeatable after loading version 1.7 , any pointers on how to proceed appreciated.

- go to HARDWARE menu and do a factory reset
- long press OK, confirm Model 01 selected, go to SETUP
- change External RF Mode OFF to PPM and the PPM output is correct on the rear connector
- click OK, Click cancel, cancel to go back to the top level, PPM output is still fine
- power off, power on, PPM output is missing although PPM is still displayed for External RF Mode OFF
- ?

ajjjjjjjj, 2022-10-05:
There is a workaround to toggle PPM output to make it work after restart: https://github.com/OpenI6X/opentx/issues/155#issuecomment-1044572241

I believe it could be broken with "PPM in" introduction, so right in version 1.0.0, because that's where logic has changed.

RobotSub, 2022-10-05:
Thank you I will play with the work around a bit more, I wasn't successful with it just now. It you ever need help pre-testing releases let me know.

ajjjjjjjj, 2022-11-22:
Hi @RobotSub, can you let me know how it behaves on this one:
[opentx-i6x-dev-ppm.zip](https://github.com/OpenI6X/opentx/files/10070735/opentx-i6x-dev-ppm.zip)

PR #311 (2022-11-24): Fixed PPM Out
Closes #155 
Closes #283
closed 2022-11-24 by ajjjjjjjj