OpenI6X/opentx #286: Unable to use FS-SM100 FMS Cable for simulation

Murugesh-Hobbyist, 2022-10-18:
Hi,
I have flashed my FSI6X TX with OpenTX v.1.7.0
Connected the FS-SM100 RC USB Flight Simulator FMS Cable with my TX.
The device is detected but the joystick doesn't work with any of the simulation apps.

I have modified the USB mode in radio setup to joystick and USB detect to Auto.
but the simulation doesn't works. 

the same cable I'm using in fsi6 with stock firmware. it works well with all the simulation apps.
Please let me know how can I fix the issue.
[image]

ajjjjjjjj, 2022-10-19:
i6X is a USB device therefore do not need FS-SM100, but straight connection cable:
https://pl.aliexpress.com/item/32969339325.html

Murugesh-Hobbyist, 2022-10-19:
The thing is, I don't have the direct cable provided in the i6x package.
so I wish to use the  FS-SM100
Please let me know how can I use it.

ajjjjjjjj, 2022-10-19:
Make your own cable, i.e.:
https://www.youtube.com/watch?v=8p419msLeyM

Murugesh-Hobbyist, 2022-10-19:
I made the cable. Now I can communicate with TX for firmware flashing.
But for simulation, the driver not seems to be working.
[image]
[image]
[image]

I also tried the following the below video, still it didn't worked out.
[https://www.youtube.com/watch?v=mRFiRYlpvoM&t=237s](url)

Please help me how can i use this cable for simulation by fixing the driver issue.

ajjjjjjjj, 2022-10-19:
There are known issues with USB Joystick, look thru other USB issues to get familiar with it. Also see: https://github.com/OpenI6X/opentx#usb-connection

Murugesh-Hobbyist, 2022-10-19:
I exactly did the same.
FlySky FS-i6X don't have a USB VBUS making it impossible to detect USB connection. Without the modification you have to go to General settings and set USB Detect to "Once" every time you're connecting USB.

The device is detected as joystick but the driver is not apt able. and I could see a square box on top left after driver update.
[image]

Murugesh-Hobbyist, 2022-10-21:
Hi Team,

I have tested the simulation by flashing the backed up stock firmware.
I can use both the data cable and simulator cable for simulation in stock firmware but not in open Tx.

When I use the stock firmware with data cable,
[image]

When I use the stock firmware with simulator cable,
[image]

But in Open TX, it doesn't working even if I change the radio setup -> Usbmode and usb detect.

Murugesh-Hobbyist, 2022-10-21:
Anyways found a solution to use the IBUS protocol in receiver, to use the Open TX firmware communication for simulation wirelessly. Thanks for the help. I'm closing the issue.

closed 2022-10-21 by Murugesh-Hobbyist