OpenI6X/opentx #299: align hardware name to left?

psicho100pro, 2022-11-12:
when i create a name in hardware for anything, then in the inputs that name interferes with the curve rendering
[image]

ajjjjjjjj, 2022-11-13:
Can you share your configuration details - where is "arm" defined to be available as a source?

psicho100pro, 2022-11-13:
It doesn't matter what resource I use, and only how I name it, if a name is assigned, its name always hits the graph
, I now have two transmitters, on one the second version with dicord and the second compilation v1.8.0 without elrs and they do the same thing.
If only the native name is , everything is fine.
[image]
[image]

ajjjjjjjj, 2022-11-13:
It does matter. If you cannot answer my question i cannot help you.

psicho100pro, 2022-11-13:
In that case I worded it wrong,,,
All the switches do it, if i assign a name to it it moves like to the right and hits the graph.
Pots do the same,,

ajjjjjjjj, 2022-11-13:
Ok, so it's a switch name, because i.e. output name works differently.

PR #314 (2022-11-29): Fix for invalid custom source name alignment
Fix for invalid custom source name alignment
port of https://github.com/EdgeTX/edgetx/pull/2538
Minor refactor
Closes #299
closed 2022-11-29 by ajjjjjjjj