OpenI6X/opentx #323: Failsafe no set warning keeps popping up

proffesionaldumbass, 2022-12-17:
When i first flashed the remote everything was working great, but after a power cycle the Failsafe not set warning came up as expected, i cleared it but then after a few seconds it popped up again and reset everything, this happens over and over so i cant set failsafe up because there is not enough time. I tried rebooting, turning on dfu mode but nothing works. I will try to reflash tomorrow but my first attempt failed so i dont have high hopes

ajjjjjjjj, 2022-12-18:
Try 1.8.0

proffesionaldumbass, 2022-12-18:
Unfortunately while flashing the display gave out, i changed it but nothing happened so the display driver is bricked. But from the sound the firmware is working allright now

ajjjjjjjj, 2022-12-18:
I'm not 100% sure but we had blank display issues before and similar reboot problems and it was faulty MCU problem.
Are you able to bind to receiver on original firmware?

Bash365, 2022-12-18:
I have the same issue

ajjjjjjjj, 2022-12-18:
@bash, good that you reported, you cannot expect something being fixed without letting know that it is broken.

Bash365, 2022-12-18:
True that. I thought I'd wait a bit to see if anyone else had the same
issue. Mine happened after a flash from original  firmware as well forgot
to mention it

ajjjjjjjj, 2022-12-20:
@proffesionaldumbass , @Bash365 , can you try on this one?
https://output.circle-artifacts.com/output/job/da2ab32e-56be-432f-8d24-e0e3b9c31609/artifacts/0/output/openi6x_en_221220.bin

Bash365, 2022-12-20:
Will try it shortly Sent from Mail for Windows From: JanekSent: Tuesday, 20 December 2022 18:12To: OpenI6X/opentxCc: Bash365; MentionSubject: Re: [OpenI6X/opentx] Failsafe not set warning keeps popping up (Issue ***@***.*** , @Bash365 , can you try on this one?https://output.circle-artifacts.com/output/job/da2ab32e-56be-432f-8d24-e0e3b9c31609/artifacts/0/output/openi6x_en_221220.bin

Bash365, 2022-12-20:
Tried it. Doesn't boot up. Screen stays blank

ajjjjjjjj, 2022-12-20:
1.8.0 is working?

Bash365, 2022-12-20:
Yes 1.8.0 works but with no proper working ppm and no afhds support

ajjjjjjjj, 2022-12-22:
If you are powering it with 2S directly there is a chance that you damaged your radio and now 1.8.0 or original software may also not work correctly, like unable to connect to receiver.

Bash365, 2022-12-22:
> If you are powering it with 2S directly there is a chance that you damaged your radio and now 1.8.0 or original software may also not work correctly, like unable to connect to receiver.

Thats 6.6v life its not a lipo. so the voltage i believe is fine. I already have stated that I have no other issues with the radio using original firmware and also 1.8.0 so i believe the issue is not with damage to my radio as it doesnt display any other issues as you have described. 
this isnt my thread as well so im not the only person with this issue.

Bash365, 2022-12-22:
Also telling me I damaged my radio seems like its the go to excuse for certain features of a aftermarket open source software. Its the software thats the issue not the hardware in this case

ajjjjjjjj, 2022-12-22:
> Thats 6.6v life its not a lipo. so the voltage i believe is fine. (...)
>Also telling me I damaged my radio seems like its the go to excuse for certain features of a aftermarket open source software. Its the software thats the issue not the hardware in this case

LiFePO4 2S fully charged is about 7.3V, well above LDO spec.
The only change that could trigger boot loop in 1.9.0 is failure of initializing additional timer - the one responsible for PPM.
This is not a valid place for your comments on opensource community.

Bash365, 2022-12-22:
Well I apologize if I hurt your feelings but there's no other forums nor is there anything else linked to this forum or related in any way where I can vent how I feel about opentx. I didnt mention the community as a whole. I did however mention I'm using a module so possibly this is the cause as it's directly related and connected to the ppm out.

Bash365, 2022-12-23:
I'll continue to wait until all the features have been implemented that I need and when all the bugs are ironed out. Until then I'm happy using original firmware. NO ISSUES AT ALL!!

ajjjjjjjj, 2022-12-23:
No need to apologize, no feelings hurt. 
> there's no other forums nor is there anything else linked to this forum or related in any way where I can vent how I feel about opentx

Main page readme links to groups on: Telegram, Discord and RCGroups. 

@proffesionaldumbass, there were similar stories in https://github.com/OpenI6X/opentx/issues/204 and https://github.com/OpenI6X/opentx/issues/246 Can you confirm your case is similar and/or you are powering from 2S lipo direclty?

proffesionaldumbass, 2022-12-23:
I was powering it off of the standart 4x aa batteries, so no issue there

PR #336 (2023-01-08): Restore heartbeat workaround
Restored heartbeat workaround because PPM init fails for some users.

Closes #323
closed 2023-01-08 by ajjjjjjjj