OpenI6X/opentx #342: Lua script implementation

eksperymento, 2023-01-28:
Is it possible to run INAV,Yapu lua scripts on this firmware?

ajjjjjjjj, 2023-01-28:
Lua is not possible on i6X.

psicho100pro, 2023-01-30:
Hi @eksperymento , maybe this is what @FRPProject is trying to do, I don't know. I'm trying something too. I've never worked in C, I'm slow and busy elsewhere. So far, the real and serious limitation is memory, it is possible that it will not even succeed. Some limited form maybe yes. So far I only have a sketch of the drawing and the mathematical basis of the artificial horizon.
I can't handle the automatic assignment of sensors with my knowledge yet.

ajjjjjjjj, 2023-01-30:
@psicho100pro , @FRPProject, if you would like to share your work as PR draft i could take a look and help to make things smaller, if you don't mind working on that together.

FRPProject, 2023-01-30:
@eksperymento @psicho100pro @ajjjjjjjj 

A few weeks back, I started with a similar idea of making the INAV widget because I was very excited about the "Hello World!" feature of OpenI6X (See: https://discord.com/channels/973289741862727741/973289742588321844/1056794523495321641). (I believe lots of people want to help out with the development of the OpenI6X project but don't know how/where to start). I think @ajjjjjjjj did a great job of triggering folks' initiative to build their own widget via the "Hello World" feature.  Thanks to @ajjjjjjjj support on [OpenI6X Discord](https://discord.gg/3vKfYNTVa2), I learned about [Docker](https://www.docker.com/) and [CircleCI](https://circleci.com/) to compile my very first hardware firmware using GitHub online editor. I am still very new to all of this but glad that I took the initiative😅

However, like many others, I found the limited memory space (and the fact that there is no Lua compiler); So, with my limited time and experience in C/(hardware programming/GitHub/...), I shortly came to the conclusion that it's best for me to at least finish a simple utility tool I wanted to have on my radio:
[image]
(Press OK-> centers the HOME position, Up/down key-> changes the plot x/y positive directions). (See: https://discord.com/channels/973289741862727741/973290217266118677/1060291095138939002)

psicho100pro, 2023-01-30:
@ajjjjjjjj I have to put my work in a form where I won't be ashamed of it and then share it. Until I really don't know what to do next.

Ultraexpert, 2023-01-30:
An inav lua flavor would be nice

PR #346 (2023-05-14): An attempt to implement custom INAV radar screen
It's not fully featured yet and not properly tested.

Closes #342 

TODO:
- [x] Rx batt on top bar,
- [x] ability to select as telemetry screen,
- [x] persist home location during leaving/entering screen again,
- [x] HDOP display for CRSF and IBUS,
- [x] throttle to every other frame, no need to keep 20fps
closed 2023-05-14 by ajjjjjjjj