OpenI6X/opentx #343: ELRS V3 tool not working

bastian2001, 2023-01-28:
**Describe the bug**
I have a Radiomaster Ranger Micro, running ELRS V3.2.0. The normal communication (gimbal movement and switches, as well as telemetry) are transmitted and received fine, but I cannot access the Express LRS tool. It shows loading forever, with many question marks in the top right.

**To Reproduce**
Steps to reproduce the behavior:
1. Attach ELRS V3.2.0 TX module
2. Start Express LRS tool
3. Nothing happens

**Expected behavior**
The normal options (TX power, packet rate, ...) should be visible. This works fine on my primary radio, Radiomaster TX16S.
 
**General information**
 - OpenI6X version 1.9.0 ELRSV3 with backlight mod
 - Radiomaster Ranger Micro
 - Battery mod: LiPo + Buck converter (set to 5.8V)

ajjjjjjjj, 2023-01-29:
What is your Max bauds in Hardware setup on both radios?

bastian2001, 2023-01-29:
Oh great, I didn't know 1.8MBaud wasn't the max, it was set to 115200 on the I6X. Now it works. Is there a reason why the Express LRS tool doesn't work with lower baud rates even though the transmission of the gimbal/switches is fine at the lower bauds?

closed 2023-01-29 by bastian2001

ajjjjjjjj, 2023-01-29:
There is a ticket in ExpressLRS repository regarding the same issue. It must be something with their serial implementation and i belive it will be improved in the future.

See:
https://github.com/ExpressLRS/ExpressLRS/issues/2010

bastian2001, 2023-01-29:
Oh ok, I suspected it to be a flaw related to the fact that the lua is baked into the firmware 🤷🏻