OpenI6X/opentx #348: USB Joystick doesn't work in Windows 10

Danulal, 2023-03-27:
I tried 1.7.0, 1.8.0 and 1.9.0 (release and self built with docker image and reccommended command)

When plugging the radio into my pc and selecting *USB Connection -> USB Joystick* on the radio windows device manager gives me `code 10, device could not be started` and it doesn't get detected anywhere.
Same result on every USB port on every windows machine I own.
However it works fine in linux (specifically garuda linux live) for some reason

All drivers are up to date and all windows updates are installed

megacct, 2023-03-30:
Just to confirm, you are using an i6x usb cable and not an i6?

ajjjjjjjj, 2023-04-01:
It is a known, not yet resolved issue. Waiting for volunteers to fix.

WOPR-Computer, 2023-04-05:
OK I am a newbie to open TX. I just installed 1.9.0heli. The OEM software worked great as a controller for sims on my WIN10 computer. Opentx is giving me a Code10 failure in the Driver. The Human USB interface has an exclamation mark on it. Driver failure to load.
I am getting ready to install a 4 in 1 module in order to fly a Blade NANO S3, with my i6x. 
I had my 450 heli setup with OEM software. I need to sim with OPENTX. I do not want to auger my 450.

WOPR-Computer, 2023-04-05:
The USB does connect to the computer the driver fails to load correctly..

WOPR-Computer, 2023-04-05:
This device cannot start. (Code 10)

{Operation Failed}
The requested operation was unsuccessful.

WOPR-Computer, 2023-04-05:
Device USB\VID_1209&PID_4F54\00000000001B was configured.

Driver Name: input.inf
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 06/21/2006
Driver Version: 10.0.19041.2486
Driver Provider: Microsoft
Driver Section: HID_Inst.NT
Driver Rank: 0xFF3202
Matching Device Id: USB\Class_03
Outranked Drivers: 
Device Updated: false
Parent Device: USB\ROOT_HUB30\5&44c1efe&0&0

WOPR-Computer, 2023-04-05:
Device USB\VID_1209&PID_4F54\00000000001B was not migrated due to partial or ambiguous match.

Last Device Instance Id: USB\VID_046D&PID_C52B\6&231ed10a&0&2
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Location Path: PCIROOT(0)#PCI(0801)#PCI(0003)#USBROOT(0)#USB(2)
Migration Rank: 0xF000FFFFFFFF0033
Present: false
Status: 0xC0000719

WOPR-Computer, 2023-04-05:
This is the failure.

WOPR-Computer, 2023-04-17:
I found a work around. I created a new model and placed it in the exterior TX with the interior tx off. I use a 4 channel uart  USB cable to PS2 connector. The uart connects to the computer as a USB device and reads the i6x ppm output. be cause it is only a 4 channel devise it does not see channels 5,6. AETR work. Curves can be set. The cable I am using is very old, I think it came with an E-Sky Helicopter transmitter.

WOPR-Computer, 2023-04-17:
The [INF Version section](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-version-section.md) should have the CLASSGUID and DriverVer directives set as follows:

The CLASSGUID directive must specify the Microsoft class GUID for HID devices. This GUID has the value {745a17a0-74d3-11d0-b6fe-00a0c90f57da}.

The DriverVer directive must have a value that has a newer date and greater version number than the value specified by the DriverVer directive in Input.inf.

The VendorXYZDevice* sections specify the hardware identifier (ID) for the vendor's HID device. The hardware ID consists of a vendor identifier (VID) and product identifier (PID). Each hardware ID for a device must have VID/PID values that are unique to the vendor and device. This ensures that the same hardware ID does not correspond to multiple names and settings
The VendorXYZDevice_Install.NT and VendorXYZDevice_Install.NT.HW sections are [INF DDInstall sections](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-ddinstall-section.md). In this example, these sections contain INF Include and Needs directives.
The Include directives reference the system-supplied Input.inf file, which contains INF sections needed to enable the USB selective suspend feature for the vendor's HID device.

The Needs directives indicate which sections from Input.inf should be processed during device installation. In this case, the HID_SelSus_Inst section is selected instead of the default HID_Inst section, which does not support selective suspend.

The VendorXYZDevice_Install.NT.Services section is an [INF DDInstall.HW section](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-ddinstall-hw-section.md). In this example, the section also contains the same values for the INF Include and Needs directives.

alccrl, 2023-05-12:
Hi.
Since this is an opensource project,
could this explanation be used for a sw fix ?
I am using a new laptop Azus Tuf Gaming and it seems to be working most of the time.
Once in a while  I get error code 10.
Tested simulators:
Velocidrone
X-plane
Uncrashed

Alfredo

alccrl, 2023-05-12:
On win 11

Motorbot, 2023-08-01:
Today I managed to install 1.10.0. and also get the driver error "device can not start (Code 10)" on my Win10 machine.
Is there already a fix for this issue?

It's quite a pity. One the one hand the RC now has a lot of new fancy functions like tons of mixers, but one the other hand for virtual flight training at the PC it's bricked.

rafolg, 2023-08-02:
> Today I managed to install 1.10.0. and also get the driver error "device can not start (Code 10)" on my Win10 machine. Is there already a fix for this issue?
>
> It's quite a pity. One the one hand the [...]

I was having a similar issue, albeit, in Windows 11 where the USB was not loading the drivers and I found the issue was my batteries were low, once I put new batteries, windows recognised the USB device as either storage or joystick - as per my selection on the RC handset

Motorbot, 2023-08-02:
Thanks a lot for your reply. Unfortunetely, the error stays the same. Before I had 5,5V at the RC and with new batteries 6,2V but the USB error is still there. When I choose the option "USB storage" the RC is recognized correctly as a storage device , but the joystick option does not work properly.
Do I have to do the USB VBUS mod to get it to work? Honestly I did not fully understand where to solder the wires. The descritption here says "USB data+ & data-  PA15", preferrably with a 1k resistor. Does that mean I have to connect both pins, data+ and data-, separately with a 1K resistor to PA15?
Sorry for my confusion.

rafolg, 2023-08-02:
[image]
D+ to D+
D- to D-
G to GND
V to PA15 via a series 1K resistor

DKuba4, 2023-11-23:
That will not solve the issue, I've already tried 2 different USB mods, none of them worked. The error is exactly the same, code 10, failed to start the device when using joystick mode, but works completly fine as a storage device. Gonna try on linux see what linux has to say.

alccrl, 2023-11-23:
"Original" (if ever exists ...) flashed back fw does correctly handle usb as joystick. I don't think it is a hw issie. I think is a sw issue.
Rc is recognized as "Flysky ....". and it works with flight simulators.

Dont rember here now exact versions and firmware I flashed back, but I saved everything on my pc, so I can give details, if useful.

Alfredo

alccrl, 2023-11-23:
Version flashed back is available on internet.

DKuba4, 2023-11-24:
I found a work around, it involves having a flight controller and a receiver (paired to the transmitter) connected to it.
Basically I installed VJoy and VJoySerialFeeder, I opened the VJSF, created a new profile, selected the protocol to MultiWii, connected my drone to the computer, selected the right COM port, and after that I just added the channels 1 by 1.
[image]

WOPR-Computer, 2023-11-24:
OK I gave up on the i6X open TX platform for flight simulation on my wiin10 computer. I went out and purchased a Radio Master open TX transmitter and had no problems setting it up as a simulation controller. This is a problem with the open TX i6x open TX build. It is not compatible with win10.

DKuba4, 2023-11-24:
It is compatible, just with some computers, my intel laptop can detect the transmitter no problem, it is indeed a driver issue originated by the OpenTX firmware, however you just went for the most expensive solution (buying a new radio). There is quite a few alternatives to buying a new radio just because the USB connection doesn't work.

1. Flashing an older version of the custom OpenTX firmware or the original FlySky I6X firmware.
2. Installing an ELRS module and just using BT.
3. Connecting an already paired drone FC to the PC via USB, downloading 2 windows apps (VJoySerialFeeder [Portable App] & VJoy [Driver]), and setting up the controller as shown in my previous post.
4. Connecting the PPM output of the port in the back of the radio to a jack cable, and just using ur microphone to capture the signals, there is an app that translates them.
5. Buying the cheapest receiver, pairing it to the radio, getting an arduino and connect it to the receiver, then writing a couple of code lines to translate the PPM inputs of the receiver to outputs for the PC.
6. Using a different operating system (Linux seems to work fine with the controller via USB).

ajjjjjjjj, 2024-10-07:
Fixed by #424

closed 2024-10-07 by ajjjjjjjj