OpenI6X/opentx #349: openi6x_1_9_0_heli does not allow DFU in Win10 "USB Error"

WOPR-Computer, 2023-04-05:
OK. I am new at open TX. I am not able to use joystick in 1.9.0 heli this is a WIN10 driver problem. I decided I would try 1.8.0  using STM32Cube Programmer. I placed the i6x in DFU mode using the trim switches. STM32Cube can not connect to  i6x. I have not tried to short R53 yet. I hope my i6x is not bricked.

Please see STM32Cube Response.
  07:54:27 : STM32CubeProgrammer API v2.13.0 | Windows-64Bits 
  07:54:41 : UR connection mode is defined with the HWrst reset mode
  07:54:41 : Error: Invalid USB port name !
  07:54:41 : Error: Establishing connection with the device failed
  07:55:09 : UR connection mode is defined with the HWrst reset mode
  07:55:09 : Error: Invalid USB port name !
  07:55:09 : Error: Establishing connection with the device failed
  07:55:59 : UR connection mode is defined with the HWrst reset mode
  07:55:59 : Error: Invalid USB port name !
  07:55:59 : Error: Establishing connection with the device failed
  07:57:38 : UR connection mode is defined with the HWrst reset mode
  07:57:38 : Error: Invalid USB port name !
  07:57:38 : Error: Establishing connection with the device failed

WOPR-Computer, 2023-04-05:
I see I did not explain I was going to re-flash the original OEM firmware. To regain my Joysticks in my sims.

WOPR-Computer, 2023-04-05:
R53 Shorted results.
08:42:53 : STM32CubeProgrammer API v2.13.0 | Windows-64Bits 
  08:43:08 : UR connection mode is defined with the HWrst reset mode
  08:43:08 : Error: Invalid USB port name !
  08:43:09 : Error: Establishing connection with the device failed

How do I fix this? I have a brick.

WOPR-Computer, 2023-04-05:
I was able to resolve this by relearning the STM32 driver in Device manager. I have now been able to re-flash the OEM software and resume Joystick sim operation.

WOPR-Computer, 2023-04-05:
USB problems need to be addressed. This makes opentxi6x very unfriendly if you need to use a simulator and very scary if the bootloader does not work after install in WIN10.

WOPR-Computer, 2023-04-15:
Looks like the wrong driver is being called:
Device USB\VID_1209&PID_4F54\00000000001B was not migrated due to partial or ambiguous match.

Last Device Instance Id: USB\VID_046D&PID_C52B\6&231ed10a&0&2
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Location Path: PCIROOT(0)#PCI(0801)#PCI(0003)#USBROOT(0)#USB(2)
Migration Rank: 0xF000FFFFFFFF0033
Present: false
Status: 0xC0000719

ajjjjjjjj, 2024-03-04:
Seems like it was related to your OS and drivers, is it solved now?

WOPR-Computer, 2024-03-04:
No!
I have abandoned the i6X and moved to a Radio Master 4 in1 tx unit. Runs full open TX no problem. i6X is a great radio with oem software. Openi6x_1_9_hei just is buggy. 
I am not a programmer so it is not worth my time. Like I  said I have moved on to a better radio.

closed 2024-10-07 by ajjjjjjjj