OpenI6X/opentx #365: Issues Trying to Get the I6X Detected by the SMT32 Programmer

Mel4221, 2023-08-03:
Hi Im currently trying to flash the OpenTX firmware to my I6X But sadly it seems that they change the processor and it requires a different programer  since i wire up everything completly as  i was supposed to exacly as show here: https://github.com/OpenI6X/opentx/raw/master/doc/flysky/dfu_jump_r53.jpg
but it did not work so i gess that based on the chip model of the SMT32 is possible that they did something different, but im not sure , either way i found some information about the manufacture which is:  https://global.geehy.com/support/apm32/?id=202
if you guys have any suggestion please let me know i will be willing to provide any picture or information

ajjjjjjjj, 2023-08-04:
There are alternatives for APM32 MCUs described in wiki:
https://github.com/OpenI6X/opentx/wiki#alternatively-flash-using-stm32-cubeprogrammergeehy-dfuprogrammerqmk-toolbox

Mel4221, 2023-08-04:
> There are alternatives for APM32 MCUs described in wiki: https://github.com/OpenI6X/opentx/wiki#alternatively-flash-using-stm32-cubeprogrammergeehy-dfuprogrammerqmk-toolbox
i will try the other ones , due to the STM32 CubeProgrammer did not work for me for some reason does not recognized it at all
thanks for answering..

Mel4221, 2023-08-04:
First Thanks a lot for answering one more time @ajjjjjjjj here is some information about my current status  on this issue.

I have fallow the process described on the wiki and also some videos on youtube on the steps of how to install OpenI6X on my radio sadly my radio has de APM32 MCU which lacks on some videos on youtube on how to properly fallow the step by step flashing process , but here are some pictures of my process: 

step1: i first bridge the pad R53 as explained in the wiki
setep2:  solder the rest of the wires acording to the picture in the wiki here is how it looks:

>My soldering skills don't look good at all :-)
[image]

step3: now i got on the computer and before i did anything i connected everything to the ST-Link device with each wire accordingly  then  i turned on the Radio which as expected did not  turn on the lights  then i connected the connector to the computer but which made a sound but here is my original issue any of the Programmers are either not detecting the  device here is some more pictures : 
>QMK_ToolBox
[image]

>APM32_Programer
[image]

>SMT32_CubeProgrammer
[image]
>SMT32_CubeProgrammer
[image]

@ajjjjjjjj  if you have any suggestion or ideas please let me know , and again thanks a lot for taking a second to answer.

Mel4221, 2023-08-05:
I have had some minimal progress 
but still nothing yet 
[image]
i even found a video on youtube with a different approuch on how to flash it(https://www.youtube.com/watch?v=ujlnielHad4 ) , but still same problem i am not getting it detected by the drivers manager 
[image]

Ultraexpert, 2023-08-05:
Uninstall the driver restart pc,check new batteries,put into dfu mode , plug in i6x into pc, open qmktoolbox
Flash. Search keenas on rcgroups I have a guide somewhere on there.

Ultraexpert, 2023-08-05:
I plan to share a video demonstrating my process with the APM32 once I find the time to upgrade to the recently released iNav version. Anticipating to have it ready by next week!

Ultraexpert, 2023-08-05:
I found the old post for you https://www.rcgroups.com/forums/showthread.php?3916435-FlySky-I6X-port-of-OpenTX/page16#post48731989

japanel, 2023-08-05:
@Mel4221 Hi, why you do t use original cable you got with your I6X? I flashed with DFUprogrammer on first try withou any problem. I had APM32 too. I soldered R53 pads only then use original cable, without issue.

Mel4221, 2023-08-05:
> @Mel4221 Hi, why you dont use original cable you got with your I6X? I flashed with DFUprogrammer on first try without any problem. I had APM32 too. I soldered R53 pads only then use original cable, without issue.

That would be aswome i was currently using the ST-Link v2 for that and i also try using the DFUProgrammer for that but not much luck on that either because is not detecting the DFU Mode status for some reason even through the pads are sordered correctly. 

But if you have a link to the installer for the DFUProgrammer that you used please shared it that i would be willing to give it a try , are  some pictures about my actual results: 
>for some reason i can not see why is not getting detected and the pad R53 is the one that is Briged so i don't know where to start if i am not getting  any signal that the device is getting detected as being in DFUMode for some reason
[image]

Thanks so much for answering @japanel

Mel4221, 2023-08-05:
> I plan to share a video demonstrating my process with the APM32 once I find the time to upgrade to the recently released iNav version. Anticipating to have it ready by next week!

@Ultraexpert Thanks a lot for all the support i am currently reading all the information that you provided ,  so i will keep you all posted about my current state and again thanks a lot for all your help @Ultraexpert

Mel4221, 2023-08-05:
I Mean i don't know how but  for some reason now when i tried again it was detected by the DFUProgramer by using the original cabel directly instead of the ST-Link
One more time thanks a lot @japanel 
[image]

japanel, 2023-08-05:
> I Mean i don't know how but for some reason now when i tried again it was detected by the DFUProgramer by using the original cabel directly instead of the ST-Link One more time thanks a lot @japanel [image]

@Mel4221 great, same version, downloaded from https://global.geehy.com/support/apm32/?id=383
I hope you are sucessfull now.

Mel4221, 2023-08-05:
I Would like to start first by saying thanks to all of you that gave me so much assitence through this process of flashing my radio , thanks a lot for being patient and also for the time inbested to help @ajjjjjjjj  @Ultraexpert  @japanel @marianomd 

So  yeah finally i was able to do it flash it , and im going to list down the issues that i strugle with the most 

1.  Wrong wiring to the ST-LINK which was my mistake entirely 
2.  Not Needed Drivers since i only needed to download the QMX Toolbox which had all the drivers that i  needed 
3.  The DFU Programer here : works on regards that it does detect the device but it did not allow me to flash it for some reason ("The flashing button did never appear for some reason")
4. I Had some random issues with the QMX Toolbox  because it was always able to detect the APM32 but it was giving me an error at flashing time later on i notice that one i lunch it from the terminal it change one Setting  to this value [image]  
which i trully have no idea what is for but it looks to be the right setting for my MCU 

**One More  Time Thanks a lot for all your help : @ajjjjjjjj  @Ultraexpert  @japanel @marianomd **

> Here is 
[image]

>I Added the wire for the Back light 
[image]

closed 2023-08-05 by Mel4221